damiafuentes/DJITelloPy #2: Most of the commands including takeoff/land returns false but still partially done by Tello.

DogukanAltay, 2019-01-08:


damiafuentes, 2019-01-08:
The library is only an interface between Python and Tello. If Tello returns True or False to a command, the library returns the same. I don't think it is a problem of the library as me and other people have tried that and worked as expected. Although that, it is true that sometimes Tello responds a little bit strange and behaves crazy. But you can try with the following:
- Is your Tello firmware up-to-date? You can do that through the Tello app. 
- Are you sending commands too fast? I send commands every half second, so Tello is not saturated.

DogukanAltay, 2019-01-08:
> The library is only an interface between Python and Tello. If Tello returns True or False to a command, the library returns the same. I don't think it is a problem of the library as me and other [...]

- Yes, it is up-to-date.
- Actually I am not I have tried to give 5 seconds delays between each command. Somehow, Tello returns "Timeout exceed on command <command>" kind of error.

Don't get me wrong though. The library is neat, I just created the issue if you or other faced similar problems since there are almost nothing about this kind of an issue.

damiafuentes, 2019-01-08:
Have you tried to increase the RESPONSE_TIMEOUT value in Tello.py class? It is set to 0.5 seconds by default. "Timeout exceed on command" kind of error is returned when any command is sent to Tello and Tello doesn't give an answer in less than RESPONSE_TIMEOUT seconds.

damiafuentes, 2019-01-08:
Can you check through the library if it is problem of the library or is Tello directly returning that? If it is Tello, I am afraid we cannot do anything to fix that. I believe the response "not joystick" is a response directly from Tello.

When I work with Tello I noticed that when the battery is low it has strange and unpredictable behaviors. So, make sure to have full battery too.

DogukanAltay, 2019-01-09:
I have checked througly. Library is not the issue. Probably my Tello has some problems about its optical flow. Thanks for the help. We can close the issue.

closed 2019-01-09 by damiafuentes