damiafuentes/DJITelloPy #4: warning: Error opening file (/build/opencv/modules/videoio/src/cap_ffmpeg_impl.hpp:901)

senglk, 2019-02-07:
I tried to run the script example.py, but the pygame window was none responsive. The flashing light from the tello turns from green to red. Further waiting results in warning messages (See below) to appear.

Printout after running the script:
```
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
Send command: command
Response: b'ok'
Send command: speed 10
Response: b'ok'
Send command: streamoff
Response: b'ok'
Send command: streamon
Response: b'ok'
warning: Error opening file (/build/opencv/modules/videoio/src/cap_ffmpeg_impl.hpp:901)
warning: udp://@0.0.0.0:11111 (/build/opencv/modules/videoio/src/cap_ffmpeg_impl.hpp:902)
```

Other information:

```
Windows 10 Home 1803, OS build 17134.523
Tello Firmware: 01.04.78.01
Numpy: 1.11.1+mkl
Opencv-python: 4.0.0.21
Pygame: 1.9.4
```

NB: The tello firmware was updated after encountering the "unknown command" error:
```
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
Send command: command
Response: b'OK'
Send command: speed 10
Response: b'OK'
Send command: streamoff
Response: b'Unkown Command!'
Command streamoff was unsuccessful. Message: Unkown Command!
Could not stop video stream```

senglk, 2019-02-07:
I managed to fix the no video issue -- simply by port forwarding UDP port 11111.

But I am seeing some video related errors like:
```
[h264 @ 0000019ec5bb3500] non-existing PPS 0 referenced
[h264 @ 0000019ec5bb3500] non-existing PPS 0 referenced
[h264 @ 0000019ec5bb3500] decode_slice_header error
[h264 @ 0000019ec5bb3500] no frame!
```

If the above is normal and to be expected, then one can consider this issue closed.

Thank you.

closed 2019-02-07 by damiafuentes

Seang95, 2019-04-10:
I just wanted to add to this since I had the same issue. It seems to say this if it cannot get any feed from port 11111. I solved it by disabling my firewall. So disabling your firewall or just allowing it through can be another solution. Hope this helps others who have this same issue.