damiafuentes/DJITelloPy #19: Velocity giving wrong, accurate position

parakh08, 2019-08-14:
Tello using the command get_speed() publishes constantly a speed of 10 irrespective of motion. Also, how is possible to get its current position or distance it has travelled?

M4GNV5, 2019-08-14:
I assume the `get_speed` command gives the speed as set by `set_speed`.

Tellos support a special protocol which transmits velocity, acceleration, height, battery and some other **sensor** readings, but it currently is not implemented in this library.

You can check [the Tello SDK manual](https://dl-cdn.ryzerobotics.com/downloads/Tello/Tello%20SDK%202.0%20User%20Guide.pdf) under "Tello State" (page 6) and the information on how to receive the state on page 2. Implementing this into DJITelloPy would be a great addition :wink:

M4GNV5, 2019-08-14:
I was able to receive Tello state information by using a simple netcat command. It does not allow you to use these parameters in your program, but are useful for debugging and manual checks.

```bash
nc -lup 8890
```

parakh08, 2019-08-15:
yeah, but I am trying to retrieve the velocity during a mission, hopefully, I will be able to send a pr for that today.

parakh08, 2019-08-15:
Pull Request #20 solves.

closed 2019-08-15 by parakh08