damiafuentes/DJITelloPy #21: Video Stream does not work

0bscur3, 2019-08-15:
Hi, 

I was trying to get the `example.py` to work, but video stream does not seem to work. I am using python 3.6.9 on Mac OS. 

This is my stdout:

```
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
Send command: command
Response: ok
Send command: speed 10
Response: ok
Send command: streamoff
Response: ok
Send command: streamon
Response: ok
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] non-existing PPS 0 referenced
[h264 @ 0x7f86968c6600] decode_slice_header error
[h264 @ 0x7f86968c6600] no frame!
[h264 @ 0x7f869801c800] error while decoding MB 48 23, bytestream -5
[h264 @ 0x7f869801fe00] error while decoding MB 30 41, bytestream -8
```

Thanks!

acdgr8, 2019-08-28:
Hi, I'm getting a similar error when trying to switch on the stream. Interestingly, I am able to extract frames out of it using the following code -
from djitellopy import Tello
import cv2
import time
tello =Tello()
tello.connect()
tello.takeoff()
tello.streamon()
count = 0
vidcap = cv2.VideoCapture(tello.get_udp_video_address())
success,image = vidcap.read()
success = True
while success:
    vidcap.set(cv2.CAP_PROP_POS_MSEC,(count*1000))    
    success,image = vidcap.read()
    print ('Read a new frame: ', success)
    cv2.imwrite("./img/frame%d.jpg" % count, image)     # save frame as JPEG file
    count = count + 5

And the frames are being stored correctly. Can someone please help out why the videostream is not being displayed in a separate window. 
Thanks in advance!

RahilRehan, 2019-09-07:
Did anyone get it working?
Same issue here, But Video stream till the starts flying. Video stream stops as soon as takeoff and doesn't respond after that.

yin221, 2019-09-11:
I have the same issue on video streaming. Does anyone have a solution to that?

RahilRehan, 2019-09-12:
> I have the same issue on video streaming. I managed to run the program but the pygame window is black and after awhile the pygame window just closes itself (managed to took off, then the window shuts) while the drone is flying and it auto lands after awhile. Can anyone help me out on that?

I got it working on Linux system. 
Note: I did not clone the repo but instead used the djitello library.
terminal: pip install djitellopy
download example file from github and try running. 
Worked for me.

duncanmix, 2019-09-25:
> I have the same issue on video streaming. I managed to run the program but the pygame window is black and after awhile the pygame window just closes itself (managed to took off, then the window shuts) while the drone is flying and it auto lands after awhile. Can anyone help me out on that?

I have the same problem, Did you solve it?

blugo1, 2019-09-27:
This code is running using a pygame window, to ensure you have a correct render in the window try and run the pygame example and see if you get a correct rendition of the game using the following command in terminal:

`$ python3 -m pygame.examples.aliens `

If you are seeing a blank screen there is a probable compatibility issue with pygame, the issue can be fixed by running the following command in terminal using:

 

`$ pip3 install pygame==2.0.0.dev3`

Run the previous command again and you should see the game render properly which shows that pygame is properly being rendered.

 

If it is properly rendered run the example.py again and you should see a proper stream from the pygame window using the tello drone.

NOTE: That the video stream might be a bit laggy at first but it will adjust itself in a few seconds.

duncanmix, 2019-09-29:
> This code is running using a pygame window, to ensure you have a correct render in the window try and run the pygame example and see if you get a correct rendition of the game using the following [...]

still not working

M4GNV5, 2019-09-29:
Maybe the problem is not with the library but rather with pygame? You could try [this example script](https://github.com/hopfenspace/TelloScripts/blob/master/camera.py) which uses OpenCV instead.

Also please post the output of the program in case it does not work.

ntamas, 2019-10-10:
FWIW, I can confirm that the video stream does not work with macOS 10.14 and the latest stable `pygame` version (1.9.6), but upgrading to 2.0.0.dev3 as suggested by @blugo1 resolves the problem, so I guess the issue is on pygame's side (and will hopefully be fixed with the 2.0.0 release).

the-lotuseater, 2019-11-24:
pygame v1.9.6 gives a blank screen, upgrading to 2.0.0 doesn't help either, gives a glitchy screen.
(system is Manjaro linux-illyria)

Walt-H, 2020-10-27:
The OpenCV version works for me great (see [here](https://github.com/hopfenspace/TelloScripts/blob/master/camera.py)), but after doing the following, I was able to use pygame and stream:
```
$ pip install av
$ pip install opencv-python
$ pip install image
$ pip install pygame
```

closed 2020-10-27 by M4GNV5

xxffrr, 2021-10-30:
> still not working

Do you sovle it? I meet the same problom