damiafuentes/DJITelloPy #23: Is the swarm control for Tello edu only?

hyansuper, 2019-08-26:
Hi, I am very interested in the swarm control, but the normal version of tello (not tello edu) require the computer to connect to the drone's own wifi at 192.168.10.1, so I wonder if your code can be used to control multiple tello drones at different IPs.

M4GNV5, 2019-08-26:
Hey,

the swarm functionality simply sends commands to multiple tellos simultaneousely. In theory it is possible to use this with regular tellos, but you would have to connect to each tello in your swarm. Thus you would need a wifi adapter for each tello and make some NAT magic to give each tello its own ip address in a final joined network.

[This serverfault](https://serverfault.com/questions/418522/is-it-possible-to-connect-2-networks-with-same-ip-range) thread seems to be somewhat related regarding NATing the tellos.

closed 2019-08-27 by hyansuper

dxenos, 2021-01-18:
@M4GNV5 , @hyansuper 
I am also interested in flying multiple normal Tello. I use a macbook and programming with python.
I tried to create VLAN in mac but it sounds complicated to me. Any ideas or references?

M4GNV5, 2021-01-18:
Can you connect to multiple WiFis at once with your Laptop? A VLAN won't be of much help. Your python program tries to connect to `192.168.10.1` which is your Tello. This works fine when you have only one Tello, but once you are connected to two Tellos at once your computer won't know which one to communicate with. As noted above using a router which does network address translation (NAT) might work.

I guess an alternative simple solution is to have multiple laptops/raspberrypis/PCs on one network (via ethernet) and each one connecting to one Tello WiFi network.
Each laptop/raspi then has one Tello reachable at `192.168.10.1` and has an IP address in the shared ethernet network (e.g. `10.0.0.xxx`. By running [UDP proxies](https://stackoverflow.com/questions/1803329/simple-udp-proxy-solution) on all computers and connecting to the Tellos should be possible through the `10.0.0.xxx` addresses.
This isn't very 'good' solution either though.

dxenos, 2021-01-18:
Thanks for the quick response. The NAT approach sounds too complicated for me (you also need to get two routers). :( 
Your idea sounds good. Thanks! The only complication I see is that I would need multiple machines with the same setup (same python version etc). I would try this if I cannot find any other easier solution...

Do you think if I use the wifi of my mobile to connect with one of the drones there is a way to send/receive information between main the laptop (which runs the main python code and connects with Tello A) and the mobile which is connected with Tello B?

dxenos, 2021-01-27:
@M4GNV5 I followed your advice, I connected two different types of tellos and they communicate through the ethernet now. 
here is an example
https://youtu.be/Vj1EqlgAq3E