damiafuentes/DJITelloPy #28: Oserror: [winerror 10048] only one usage of each socket address (protocol/network address/port) is normally permitted

projectdrone2, 2019-09-30:
Damià, great job! Thanks! I just observed one issue. I can only run my script on Spyder once. The second time I get the following error "Oserror: [winerror 10048] only one usage of each socket address (protocol/network address/port) is normally permitted". I thought tello.end() would close nicely the socket but no. There is a workaround (in cmd.exe, run "netstat -ano|findstr 8889" then "taskkill /F /PID [ProcessID]") but I have to do it manually every time which is not so convenient. DO you have the same issue?

M4GNV5, 2020-10-27:
Hey,

sorry for the late response. I assume your issue is caused by DJITelloPy not closing the receiving sockets. On Linux I had similar problems with other programs, but afaik only with TCP sockets.

I added above commit to a seperate branch. In theory it should fix your problem by closing sockets after the last Tello object is deleted. But it may induce a delay of ~1s until the program exits. In general i do not really like the way this is implemented, but there do not seem to be many alternative ways on implementing this.

Can you clone the [socket-close](https://github.com/damiafuentes/DJITelloPy/tree/socket-close) branch and tell me if it fixes your issue?

Walt-H, 2021-01-01:
This happens to me a lot with PyCharm: after running the script once, I can't immediately run it a second time.
I usually just try to run it again (3-4 times), then it works.

**However**, I am almost always able to run a second time immediately when my code gracefully terminates:
```
tello.land()
tello.end()
```

M4GNV5, 2021-01-01:
@Walt-H Does the land+end fix only work with the code from the socket-close branch or also with the code from master?

closed 2021-02-14 by M4GNV5

tpt5cu, 2021-04-26:
if you close your terminal, it should close the socket