damiafuentes/DJITelloPy #30: In swarm scenario, how to give separate commands to tellos?

Shatad-Purohit, 2019-11-19:
Hello,

Trying to give separate commands to tello UAVs in swarm. Want to implement following scenario
-1'st UAV Takes off
-1'st UAV Lands
-2'st UAV Takes off
-3'st UAV Takes off
-2'st UAV Lands
-3'st UAV Lands

Can you help?
Want to try further condition based scenarios (stop/land based on battery level/other UAV status, etc)

Following code gives an error, also tried customizing swarm.py, tello.py but no luck.

Code:

from TelloSDKPy.djitellopy import TelloSwarm

swarm = TelloSwarm.fromIps([
    "192.168.1.45",
    "192.168.1.46",
    "192.168.1.47"
])

swarm[0].connect()
swarm[1].connect()
swarm[2].connect()

swarm[0].takeoff()
swarm[0].land()

swarm[1].takeoff()
swarm[1].takeoff()

swarm[2].land()
swarm[2].land()

swarm[0].end()
swarm[1].end()
swarm[2].end()

UltraBob, 2019-11-19:
For swarm [1] you have two takeoff commands, and for swarm [2] 2 land
commands.

Shatad-Purohit, 2019-11-20:
I updated the question, the real hurdle is only one usage of socket address is permitted.

I get following error.

Traceback (most recent call last):
  File "telloswtest.py", line 6, in <module>
    "192.168.1.45"
  File "C:\Users\shata\PycharmProjects\tellotestpy3\TelloSDKPy\djitellopy\swarm.py", line 24, in fromIps
    enable_exceptions=enable_exceptions
  File "C:\Users\shata\PycharmProjects\tellotestpy3\TelloSDKPy\djitellopy\tello.py", line 92, in __init__
    self.stateSocket.bind(('', self.STATE_UDP_PORT))  # for accessing the states of Tello
OSError: [WinError 10048] Only one usage of each socket address (protocol/network address/port) is normally permitted

M4GNV5, 2019-11-20:
Hey, i fixed that error in my fork and opened a pull request. Unfortunately damian pushed some changes to master since then.
For now you can use my fork and it should work, on the long run the conflicts should be resolved and the fixes should merged into master.

https://github.com/M4GNV5/DJITelloPy/tree/new-networking
(note the branch, which might require you to do an additional `git checkout`)

M4GNV5, 2020-05-20:
fixed by 143fe76c4280fbed174d5878231e68db66fe3a7d and 552edd1fab4a4c5508ee8740c029fbb553e80d9a

closed 2020-05-20 by M4GNV5