damiafuentes/DJITelloPy #31: All tello drone functions work perfectly except takeoff.

AditMeh, 2019-11-30:
I just started using this library with my new tello drone and I'm quite happy with how well it's working. However, out of all the methods I can call for my tello drone, the only one that doesn't work properly is the drone.takeoff() method. Whenever I call this method, my drone takes off but then after 15-20 seconds it just lands back on the ground. This is the output I get when I call the method:
```Send command: takeoff
Timeout exceed on command takeoff
Command takeoff was unsuccessful. Message: False
```
This bug has occurred when my drone is fully charged and updated to the latest firmware. Is there a workaround/fix to this issue?

M4GNV5, 2019-12-01:
When the tello takes off it orients itself using its cameras and sensors for stable air positioning.
Depending on the light and surrounding this can take a few seconds. You can try with better light conditions or taking off at a different place.

The library waits for commands 7 seconds by default, alternatively you could change that number:
https://github.com/damiafuentes/DJITelloPy/blob/master/djitellopy/tello.py#L19

M4GNV5, 2020-05-19:
fixed by #41 (longer timeout for `takeoff`)

closed 2020-05-19 by M4GNV5