damiafuentes/DJITelloPy #32: Can't install on windows10, Ubuntu

LordHandLee, 2019-12-12:
As of December 11, 2019, this package cannot be installed on any current platform. I have tried every solution as per google and still nothing works. I even tried copying and pasting the parts of your code I need in particular(procerssing UDP steam in opencv in a seperate thread) and still nothing works. I have tried experimenting with sections of your code and I cannot seem to get anything to work as far as processing UDP video. I can stream UDP through Opencv2 videocapture by itself and it works fine. When I do the same in a thread, it comes in choppy but still relevant. However, when I apply computer vision, it becomes unmanageable with an almost indefinite delay. I am assuming you have figured out the solution without being able to run your code. Could you shed some light on the process behind processing UDP via python and opencv. I am sure once I delve deep enough and study UDP, networking, IT, and opencv then the answer will become apparent. Until then, could you enlighten the issue?

M4GNV5, 2019-12-12:
Hey,
can you post some code and/or error messages you are getting?
I assume you are reading from the camera one frame after the other when you should instead always use the newest frame for your CV code. That's why the `BackgroundFrameRead` exists.  Your code should look something like this:

```python
tello = Tello()
tello.connect()
tello.streamon()
frame_read = tello.get_frame_read()

while True:
    img = frame_read.frame
    # do some CV with img here...
    # control the tello based on the CV results...
```

LordHandLee, 2019-12-13:
It does not install on Windows 10 because it requires opencv-python which overrides/blocks the opencv-contrib-python installation which contains crucial functionality I need for computer vision. 
    I am able to follow the directions for installation on Ubuntu up until I get to `$ pip install -r requirements.txt` and then I get the following error: `Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip import main
ImportError: cannot import name main
`
When I type `pip3 install -r requirements.txt` I am able to install the dependencies, but then when I run a simple script to connect, I get the following error:
`Traceback (most recent call last):
  File "takeoff.py", line 1, in <module>
    from TelloSDKPy.djitellopy import Tello
ImportError: No module named TelloSDKPy.djitellopy
`
Also, could you enlighten me as to why UDP needs to be processed in a separate thread and why it locks up the main one? Many thanks.

M4GNV5, 2019-12-16:
> $ pip install -r requirements.txt and then I get the following error: Traceback (most recent call last): File "/usr/bin/pip", line 9, in <module> from pip import main ImportError: cannot import name main 

That looks like a problem with your python2 installation.

> Traceback (most recent call last): File "takeoff.py", line 1, in <module> from TelloSDKPy.djitellopy import Tello ImportError: No module named TelloSDKPy.djitellopy 

try changing line 1 of the example to `from djitellopy import Tello`

LordHandLee, 2019-12-20:
I managed to get it installed by leaving out the first line in the following:
`$ pip install --upgrade pip
$ git clone https://github.com/damiafuentes/TelloSDKPy.git
$ cd TelloSDKPy
$ pip install -r requirements.txt`
If I try `pip install --upgrade pip` then when I get to the final step, I get another ImportError

However, the video does not seem to work. This make me think it could be something related with OpenCV. I have the latest version of python2, python3, and OpenCV installed. Not sure what the issue is.

closed 2020-11-06 by M4GNV5