damiafuentes/DJITelloPy #37: Commands all respond with error.

vec123, 2020-03-29:
Hi 
I cloned this repo and am running this small test script. 
`#!/usr/bin/env python

#include "ros/ros.h"
#include "std_msgs/String.h"
#include <sstream>

import rospy
from djitellopy import Tello
import cv2
import time

rospy.init_node('dji_tellopy', anonymous=False)

tello = Tello()

connected=tello.connect()
if(connected==True):
    print("connected!")
if(connected==False):
    print("Not connected!")

now = rospy.Time.now()
seconds=rospy.Time.now()

tello.takeoff()
print("took off")

tello.move_left(10)
print("moved left")

tello.rotate_counter_clockwise(45)
print("rotated")

tello.land()
tello.end()
`

M4GNV5, 2020-03-29:
`takeoff` usually fails when the battery is low. You could try to add a
```py
print(tello.get_battery())
```
right after your call to `tello.connect`

vec123, 2020-03-30:
Hi, thanks for your respondse. I tried that out, and indeed, it seems my battery was not high enough for a take-off. After loading my drone, it takes off. However all messages still respond with an error and the move_left command is not executed.... here is my console output.

M4GNV5, 2020-03-31:
The drone should control its position itself, but it sounds like your drone needs calibration. You can calibrate it using the official app (it asks you to remove all propellers and put the drone on each side for a few seconds).
Aditionally as the drone flight stability is mainly based on camera it might help to turn on the light.

How long does the takeoff command take until the error appears? Usually when the tello cant orient itself to a stable flight position the takeoff command takes longer to complete. The djitellopy library has a command timeout of [7 seconds](https://github.com/damiafuentes/DJITelloPy/blob/master/djitellopy/tello.py#L19) which might not be enough for the takeoff command, especially in dark environments and/or bad calibration.

it seems like you are calling `tello.move_left(10)` for 10cm, the minimal allowed distance is 20cm, thus it is out of range. Try `tello.move_left(20)`.

The `ccw 45` command returns `ok`, so apart from the above everything looks fine.

smorad, 2020-04-28:
Great library, thanks for releasing this!

I'm also seeing an issue with rotation commands failing in the same way. Do you know what causes these failures? They're intermittent in that they succeed maybe 20% of the time

```
tello.py - 186 - Send command: ccw 30
I0428 15:20:42.096289 140313620674304 tello.py:186] Send command: ccw 30
tello.py - 203 - Response ccw 30: error Auto land
I0428 15:20:42.162065 140313620674304 tello.py:203] Response ccw 30: error Auto land
tello.py - 186 - Send command: ccw 30
I0428 15:20:42.171342 140313620674304 tello.py:186] Send command: ccw 30
tello.py - 203 - Response ccw 30: error Auto land
I0428 15:20:42.204935 140313620674304 tello.py:203] Response ccw 30: error Auto land
tello.py - 186 - Send command: ccw 30
I0428 15:20:42.214517 140313620674304 tello.py:186] Send command: ccw 30
tello.py - 203 - Response ccw 30: error Auto land
```

Could this be caused by bad lighting/calibration? Or would that only cause instability/wandering?

lucasg1, 2020-08-25:
I also had these problems, I had to do some simple steps that we sometimes forget, make sure to check these every time you're starting a new flight:
- **Check if your battery is fully charged before the flight** (maybe tello responds differently for different battery levels)
- **Do routine calibrations with the Tello App** (if you do flights every day, calibrate every day)
- **Make sure that there is enough ambient light in your environment, if there isn't and if your floor is very uniform try to put some pillows and books in the ground** so the drone can stabilize itself using the visual stabilization that the drone has
- If the drone is hot, maybe it won't be able to fly. **Let the drone rest for a while after making a lot of flights.**

Obs: I think these observations should be in the main page, should save a lot of time for new pilots...

closed 2020-10-27 by M4GNV5