damiafuentes/DJITelloPy #39: The flight status of drone.

ch-MEIJIE, 2020-04-12:
I am not sure if the problem appeared because of my drone's hardware.

Compared to use the official app to control the drone, the drone flew lower after taking off when used this package for controlling. And the flight status was also not stable, the drone would fly to an unexpectable direction when hovering after taking off, even though no command was given. But the problem seems not existed when using the official app.

What's more, I noticed a firmware update was pushed by the official app, and I updated it before the test mentioned above. The flight control by this package worked well when Dec 2019, and I finished my experiment smoothly. But today, it did not work well as before. Is this firmware update has changed something internal and caused this problem?

M4GNV5, 2020-04-13:
Have you tried calibrating your drone? Your problems sound like the one [in this issue](https://github.com/damiafuentes/DJITelloPy/issues/37#issuecomment-606509274). If the issue still perists you probably have to contact the Ryze support.

ch-MEIJIE, 2020-04-17:
Thanks for your response, I have contact Ryze support. The feedback is that the drone flight status was determined mainly by the downward sensor, it would be affected by the light and the smoothness of the floor which should avoid reflecting. When I flew my drone last time, it was at my experiment room, which had carpet covering the floor, but THIS time, I flew my drone below the clean floor tile, which leads to the issue.

M4GNV5, 2020-04-18:
Interesting. But i guess it makes sense, their CV algorithm can probably better track points on a carpet than on a clear floor.
I guess we can close this issue now?

closed 2020-04-18 by ch-MEIJIE