damiafuentes/DJITelloPy #40: No module named 'djitellopy.decorators'.

Grubchubgrub, 2020-05-12:
Hi! I am running into the following issue when I try running example.py. 

Traceback (most recent call last):
  File "example.py", line 1, in <module>
    from djitellopy import Tello
  File "/mnt/c/Users/Anubhav/Documents/Tello/TelloSDKPy/djitellopy/__init__.py", line 1, in <module>
    from .tello import Tello, BackgroundFrameRead
  File "/mnt/c/Users/Anubhav/Documents/Tello/TelloSDKPy/djitellopy/tello.py", line 8, in <module>
    from .decorators import accepts
ModuleNotFoundError: No module named 'djitellopy.decorators'

Do you know what can be causing this? I am on python version 3.7.

Grubchubgrub, 2020-05-12:
It looks like the folder is missing the decorators.py folder that was present in earlier versions. Is this a mistake?

M4GNV5, 2020-05-12:
Sorry, that one was on me. I replaced the old decorators object in favor of python 3.5 function parameter/return typing in cf29c181bb14604f1809042755b78b86fc300f10.
I forgot the import of the removed module, but it still worked for me, because a decorators.pyc was still present on my computer.

Thanks for reporting :smile:

commit 701e3935 (2020-05-12): remove the old decorators import and return type definition of get_frame_read (fix #40)
closed 2020-05-12 by M4GNV5