damiafuentes/DJITelloPy #42: IMU error

Grubchubgrub, 2020-05-17:
Heyo. I'm running into an error regarding the imu when trying to control the drone using examply.py.
The drone is able to obtain the video feed and takeoff however it does not respond to any other commands. This is the output I am getting : 

[image]

Grubchubgrub, 2020-05-17:
So I followed the instructions on this forum (the last post in which I ran the command:
 >>rc a b c d
I no longer have the imu error but am running into a new issue. Upon takeoff the drone is able to receive commands for a few seconds but after that becomes unresponsive to commands and stops streaming video. It only lands upon killing the program.

Grubchubgrub, 2020-05-19:
Tried an older version of the library and tested multiple times, looks like the problem might be with the newer version that removes the pygame dependency in the example.py file. I'm not running into the issue when I use the older version.

M4GNV5, 2020-05-19:
Are you sure this happens since the pygame removal? The pygame dep was only used in example.py, that commit changed nothing in the actual tello library.
But maybe the code changes in 8894b537c694339ca8567737347aafb0c5239099 cause some trouble? They were submitted by @ovicus in #41 

Using the current master branch (4fe04c26579cce280059c9872e5cce350628a430) and a simple test script everything works fine for me with my Tello EDUs.
Could you give more information on your python version and Tello firmware version?

Grubchubgrub, 2020-05-20:
Yep - I am running Python 3.7 and Tello firmware 01.04.91.01

ovicus, 2020-05-21:
I tested changes on a real Tello before pushing and everything worked fine. The error you reported seems to be related to your specific drone, probably firmware version. Have you connected the drone with the mobile app to check for firmware updates?

Gautam-C, 2020-10-21:
I found the reason for this error generally comes from a lack of light when flying. At least that's what I experienced. Try turning on some lights.

closed 2020-10-27 by M4GNV5

maheerrizwan24, 2022-01-27:
> I found the reason for this error generally comes from a lack of light when flying. At least that's what I experienced. Try turning on some lights.

I faced the same problem, but the drone is up to date (firmware) and the room is well lit. Also, the drone is calibrated. It only takes off and doesn't execute the other commands like going forward or to the right.