damiafuentes/DJITelloPy #43: Time Between Commands

ghost-ronin, 2020-05-20:
Hi @damiafuentes .I working with Tello for object detection and I have some questions to you.Firstly ,your code has time between rc commands =0.5 .Is it necessary ?Has offical tello specs this time interval ? Can I send command 0.1 second???

M4GNV5, 2020-05-22:
Hey,

suprisingly in the official tello code they have no time between commands at all for the regular control commands. They send the next command immediately after a response to the previous one.

They don't have an example for the RC command, so no idea about that one. Anyways i reduces the time between commands to `0.1` for regular commands (they usually take longer than that to complete so that doesnt matter anyways) and to `0.001` for RC commands.

Using this with the pygame example i experienced noticable less input lag than with the old settings. So this should also help you with your project.

closed 2020-05-22 by M4GNV5

ghost-ronin, 2020-05-22:
Thanks Jakob.I will look at this.😉

Jakob Löw <notifications@github.com>, 22 May 2020 Cum, 13:07 tarihinde şunu
yazdı: