damiafuentes/DJITelloPy #44: Process finished with exit code -1073741819 (0xC0000005)

viet2411, 2020-05-21:
Hello,
example.py worked fine and nothing happened, but when I pressed Esc key to stop the program, the program returned the message "Process finished with exit code -1073741819 (0xC0000005)", and I can't understand why this occured.
I'm trying to record a video with opencv using this code, but this problem happened. Can you explain for me? 
Thank you and best regards.

M4GNV5, 2020-05-21:
In order to stop the example press Ctrl+C in the terminal. It will trigger an interrupt, land the drone and then exit. I guess i can add some more code to the example to make sure it doesn't exit with a non-zero exit code after landing.

For recording a video you can use `cv2.VideoWriter`, see for example: https://stackoverflow.com/a/16705188/4114905

Please note that the example is very basic and the video freezes while control commands are executed, you should probably put the video creation in a seperate thread.

M4GNV5, 2020-05-22:
I added some example code showing how to create a video file from the video stream: [record-video.py](https://github.com/damiafuentes/DJITelloPy/blob/master/examples/record-video.py)

viet2411, 2020-05-22:
Thank you for your comment. It really helps me a lot

closed 2020-05-22 by viet2411