damiafuentes/DJITelloPy #47: go_xyz_speed need 7 parameters

iemre98, 2020-07-30:
When I try to use go_xyz_speed with x y z and speed he asked me 3 more parameters. When I check in definition it says x1,y1,z1,x2,y2,z2,speed
There should be a mistake

M4GNV5, 2020-07-30:
maybe you are mixing up `curve_xyz_speed` with `go_xyz_speed`? The `go_...` version only [takes 4 parameter](https://github.com/damiafuentes/DJITelloPy/blob/master/djitellopy/tello.py#L628)

Can you post your code that fails and the full error message?

iemre98, 2020-07-31:
Yeah I check that but as you can see from the error something is wrong
[image]
[image]

M4GNV5, 2020-07-31:
It looks like you are using an old version of the library. In fact the function was wrong in an old version (it was defined twice), but i fixed that in https://github.com/damiafuentes/DJITelloPy/commit/c887f734ba6040e60da9f10044a0c70744d944c0.

I assume you installed djitellopy using pip?
Try
```sh
pip3 uninstall djitellopy
pip3 install https://github.com/damiafuentes/DJITelloPy/archive/master.zip
```

iemre98, 2020-07-31:
Yes, I did use pip. Thanks for your reply!

iemre98, 2020-07-31:
> Furthermore the code in your first screenshot looks wrong. You have to call `tello.takeoff()` before `tello.go_xyz_speed()`. Also you should use `tello.get_frame_read` instead of creating a `cv2.VideoCapture` youself. See the [take-picture.py example](https://github.com/damiafuentes/DJITelloPy/blob/master/examples/take-picture.py) or the [manual-control-opencv.py example](https://github.com/damiafuentes/DJITelloPy/blob/master/examples/manual-control-opencv.py)

Yeah, I was just trying to show the error I get before flying the Tello :)

closed 2020-07-31 by iemre98