damiafuentes/DJITelloPy #52: enforce_types has broken the package

mannykoum, 2020-08-27:
When I am trying to connect to the drone I get the following error:
```python
>>> from djitellopy import Tello
>>> tello = Tello()
>>> tello.connect()
Traceback (most recent call last):
File "<input>", line 1, in <module>
  tello.connect()
File "/home/mannykoum/Manos_repos/drone-env/dronenv/lib/python3.6/site-packages/djitellopy
  /enforce_types.py", line 42, in wrapper
  return func(*args, **kwargs)
File "/home/mannykoum/Manos_repos/drone-env/dronenv/lib/python3.6/site-packages/djitellopy
  /tello.py", line 517, in connect
  self.send_control_command("command")
File "/home/mannykoum/Manos_repos/drone-env/dronenv/lib/python3.6/site-packages/djitellopy
  /enforce_types.py", line 41, in wrapper
  check_types(spec, *args, **kwargs)
File "/home/mannykoum/Manos_repos/drone-env/dronenv/lib/python3.6/site-packages/djitellopy
  /enforce_types.py", line 19, in check_types
  if isinstance(type_hint, typing._SpecialForm):
AttributeError: module 'typing' has no attribute '_SpecialForm'
```

M4GNV5, 2020-08-27:
Hey,

i changed the code to not use `typing._SpecialForm` which does not seem to be available on <= 3.6

Also it seems like class field type hints (added in 024d495cd3617573f165ef6ecffc835271580ce1 and 6a7b20f835856af01d861297a32fa45c685019c6) are not supported by python 3.5, which means Debian 9 "stretch" is not supported by DJITelloPy.
Did you have any problems with the field typings in python 3.6?

closed 2020-08-27 by M4GNV5

mannykoum, 2020-08-31:
It's all fixed now, thanks! The field type checks work!

13096676951x, 2022-04-21:
Hello, I have encountered this problem when I run the code, can you tell me how you solve this problem? Very grateful for your help

M4GNV5, 2022-04-21:
What python version are you using @13096676951x? Its probably best to open a new issue with a more detailed description of your problem.

commit 752b81ba (2020-08-27): make enforce_types also work on python 3.6 (fixes #52)