damiafuentes/DJITelloPy #53: New Feature - GUI

sahmed007, 2020-08-29:
Proposal for GUI-based control of Tello where the user can perhaps select pre-made flight patterns (such as flying in a triangle, square, etc.) or create a new custom pattern for launch. More advanced patterns can be considered, along with buttons to safely land the Tello, etc. Video stream to GUI would also be great.

Walt-H, 2020-10-28:
You should checkout the drone blocks project: https://github.com/dbaldwin/DroneBlocks-Web

They provide a web GUI for chaining basic drone commands together, where even have an app for it: https://apps.apple.com/us/app/droneblocks/id1045826508

Check out there promo video and stuff: https://learn.droneblocks.io/

M4GNV5, 2021-02-14:
I'll close this here. But feel free to implement this feature and send a pull request or rather, as of this features size, implement it in a new repository which simply uses this library.

closed 2021-02-14 by M4GNV5