damiafuentes/DJITelloPy #54: New Feature - Enhanced Navigation Maneuvers

sahmed007, 2020-08-29:
Proposal for enhanced navigation features such as:
- collision detection which would force the Tello to move in a safer direction 
- obstacle detection which would allow the Tello to prevent moving in its intended path if something obstructs its flight path
- swarm coordination preventing two drones from colliding with one another
- etc.

If anyone hasn't already started, I would be more than willing to start implementing some of these.

Walt-H, 2020-10-28:
Each one of those cases can be a large project in themselves, where due to the lack of sensors on the Tello, would be crazy ambitious. Also, unless you have a bad ass computer or specialized hardware, you'll be operating at about 1-3 FPS.

I recommend that you start a project based on one of these ideas using this library as your client :D

M4GNV5, 2021-02-14:
I'll close this issue. But feel free to implement this feature and send a pull request.

closed 2021-02-14 by M4GNV5