damiafuentes/DJITelloPy #55: Increasing sampling rate of state

wagenaartje, 2020-10-05:
I'm using `tello.get_acceleration_z()` to get the acceleration, but I notice that the value that I receive is basically only updated at 10 Hz (even though I call the function more often). Is there a way to increase the sampling rate of the state?

M4GNV5, 2020-10-05:
The acceleration is obtained from tello state packts. These are sent by the tello without requesting them. Thus i am afraid there is no control over their frequency. It is locked to 10Hz, see [Tello SDK Docs](https://dl-cdn.ryzerobotics.com/downloads/Tello/Tello%20SDK%202.0%20User%20Guide.pdf) page 7.

wagenaartje, 2020-10-05:
What you say is indeed true, the SDK does not give any way to increase the frequency. However, I have noticed that there is more data transferred than specified in the SDK. You can see this as follows:

1. Do a flight using the official 'Tello' iOS app
2. Connect your iPhone/iPad to iTunes. In iTunes, click on your device and go to 'File sharing'. It should say Tello there, and you can download flight logs
3. These flight logs are `.dat` files. You can open them with [CsvView](https://datfile.net/CsvView/downloads.html) and then export them to `.csv`

If you open this csv in excel and separate the data by commas, you will see way more data - **over 100 columns of data!** Unfortunately, this date is also sampled at 10 Hz. But since there is a lot going on 'under the hood', i'm sure that we can increase the sampling rate as well.

M4GNV5, 2020-10-05:
sounds interesting. Can you provide one such csv file?
Afaik the tello app uses a different protocol, not the official SDK.

wagenaartje, 2020-10-06:
[This](https://github.com/damiafuentes/DJITelloPy/files/5332132/log.zip) zip file contains the raw `.dat` file and the converted `.csv` file. It would be interesting to build a python library that incorporates this data as well. If i'm correct, the library implemented in Go actually uses some of this data (https://godoc.org/gobot.io/x/gobot/platforms/dji/tello)

M4GNV5, 2020-10-27:
The files look interesting. But none of those are available via the official UDP API.

I assume these can be retrieved using the internal protocol used by the App. There is [this documentation of the protocol](https://tellopilots.com/wiki/protocol/). And there is [this repository which seems to implement it in python](https://github.com/hanyazou/TelloPy).

closed 2020-10-27 by M4GNV5