damiafuentes/DJITelloPy #56: How to install on Nvidia Jetson Nano?

djhnsn, 2020-10-26:
Installation fails, and it tries to install a version of open cv that I do not need.

M4GNV5, 2020-10-26:
can you provide some Information on the error messages you receive? What OpenCV Version do you have installed? How did you install it?

djhnsn, 2020-10-26:
When it's installing, first it identifies that I meet the prerequisite for numpy.
Then it downloads opencv-python and it fails to build the pip wheel for open cv.

I am using opencv 4.1.1 iirc.
I don't remember exactly how I installed it, but I installed it from one of the many nvidia guides as a prerequisite to something else.
I have hundreds of python programs that import it, and it works.

M4GNV5, 2020-10-26:
DJITelloPy has opencv `3.4.3.18` in its requirenments file. It could work with 4.1.1 but i am unsure how much API changes they did. You could try cloning DJITelloPy and modifying the required version in `requirements.txt` setting it to `4.1.1.`

Or you could try fixing the pip build of version `3.4.3.18`. Could you provide the full log of pip? You probably just need to install some build utils or a required native library in order for it to succeed the build.

djhnsn, 2020-10-26:
I tried to remove open cv from requirements.txt, but it still attempted the install anyway. Also, I am pretty sure the package name from nvidia isn't called opencv-python. Anyway, I will try changing it to 4.1.1. and running it again later. 

If it doesn't work, I will flash a new sd card image to avoid breaking my open cv installation and try from scratch. I will let you know.

Thanks.

djhnsn, 2020-10-27:
Alright, I changed the version number in requirements.txt, removed it, and changed to package name to just opencv == 4.1.1. It still attempted to download opencv-python. I don't have a log because I cancelled it. The jetson nano cpu is pretty slow. pip is up to date. 

Upon doing some research. I never installed open cv at all, it comes compiled for this hardware and included in the nvidia jetpack 4.4. build. Numpy was not, and was installed my me. Maybe this is why pip recognizes that numpy is installed and not opencv? 

It's unlikely that I will have success with a clean installation of jetpack 4.4, but I will give it a try sometime this week.

M4GNV5, 2020-10-27:
After you removed opencv from the requirenments you have to install the package in developer mode using `pip install -e .`. If you run the regular install command pip will just download the original source code again and ignore your changes.

Try something like:
```sh
git clone https://github.com/damiafuentes/TelloSDKPy.git
cd TelloSDKPy
# --- remove the opencv== line from ./requirenments.txt
pip install -e .
```

djhnsn, 2020-10-27:
Very good, thank you, I will try this soon.

M4GNV5, 2020-10-27:
Ohh, i just realised you may also have to remove the dependency from the [setup.py file here](https://github.com/damiafuentes/DJITelloPy/blob/master/setup.py#L17)

djhnsn, 2020-10-27:
Cool, I have a feeling that might do it. 
I hope so anyway. I will report my findings ASAP. 

Thanks again!