damiafuentes/DJITelloPy #58: enforce_types again?

Xendi, 2020-10-31:
I have basic commands (connect, takeoff, land, moves) working in WinPython 3.8.2. However, if I do any state queries I get something like the following:

tello.get_battery()

---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
<ipython-input-26-e5d6f7559881> in <module>
----> 1 tello.get_battery()

D:\Users\Dave\WPy64-3850\python-3.8.5.amd64\lib\site-packages\djitellopy\enforce_types.py in wrapper(*args, **kwargs)
     43         def wrapper(*args, **kwargs):
     44             check_types(spec, *args, **kwargs)
---> 45             return func(*args, **kwargs)
     46 
     47         return wrapper

D:\Users\Dave\WPy64-3850\python-3.8.5.amd64\lib\site-packages\djitellopy\tello.py in get_battery(self)
    374             int: 0-100
    375         """
--> 376         return self.get_state_field('bat')
    377 
    378     def get_udp_video_address(self) -> str:

D:\Users\Dave\WPy64-3850\python-3.8.5.amd64\lib\site-packages\djitellopy\enforce_types.py in wrapper(*args, **kwargs)
     43         def wrapper(*args, **kwargs):
     44             check_types(spec, *args, **kwargs)
---> 45             return func(*args, **kwargs)
     46 
     47         return wrapper

D:\Users\Dave\WPy64-3850\python-3.8.5.amd64\lib\site-packages\djitellopy\tello.py in get_state_field(self, key)
    219             return state[key]
    220         else:
--> 221             raise Exception('Could not get state property ' + key)
    222 
    223     def get_mission_pad_id(self) -> int:

Exception: Could not get state property bat

Thanks for your help!

M4GNV5, 2020-10-31:
Hey,

your exception is raised in [get_state_field](https://github.com/damiafuentes/DJITelloPy/blob/master/djitellopy/tello.py#L221) which is called by [get_battery](https://github.com/damiafuentes/DJITelloPy/blob/master/djitellopy/tello.py#L376). The wrapper function in between are inserted by the enforce_types utility, but they do not cause the problem.

Can you provide your full program? The state functions (e.g. `get_battery`) gain their values via UDP packets from the drone. The Tello starts to send these packets after you called `tello.connect()`. Thus please make sure you run `tello.connect()` before calling `tello.get_battery()`. You also might need to wait a few milliseconds between the call to `connect` and to `get_battery` to give the drone some time to send the first state packet.

Here is some example code:
```python
import time
from djitellopy import Tello

tello = Tello()

tello.connect()
time.sleep(0.5) # wait 0.5 seconds

print("Battery status = ", tello.get_battery())
```

Xendi, 2020-10-31:
I'm just using the 'Simple Example' code from the Readme, which works. If I add "tello.get_battery" at any point after tello.connect, I get the error as posted and the connection drops. Same for other state operations.

M4GNV5, 2020-11-02:
Did you try adding a `time.sleep(0.5)` between the connect and the get_battery?
It sounds like DJITelloPy does not receive any state packets from your Tello. Do you have some sort of firewall that could be blocking UDP packets (port 8890)?
Did you try updating your Tello via the official App?

commit a5652070 (2020-12-26): wait until the first state packet is received before returning from Tello.connect (fixes #58)
closed 2020-12-26 by M4GNV5