damiafuentes/DJITelloPy #59: Is it possible to obtain the states (vgx,vgy,vgz) of Tello without a mission pad?

yorklyb, 2020-11-06:
Hi, I'm using an origin Tello, not an EDU. I installed djitellopy via Pycharm, then I found that the site package was actually different from the one shown in this repository  (tello.py and other files). I'm trying to implement sensor fusion of camera and IMU, using Apriltags. I have successfully got the accelerations in XYZ and angular accelerations as well.  It seems to be impossible to get the vgx, vgy, and vgz based the tello.py. However, since we can use some commands to set velocity and let the drone move a certain distance(cm) towards the desired direction, also based on the SDK file given by DJI, I believe all the states are actually available. So I'm wondering how to get vgx, vgy, and vgz without a mission pad.

M4GNV5, 2020-11-06:
Hey,

yes the `vgx, vgy, vgz` state values are available on regular tellos. Please make sure you have the newest djitellopy installed. The version on pip is outdated. Try installing it using `pip install https://github.com/damiafuentes/DJITelloPy/archive/master.zip`

The speed values are availble through the [tello.get_speed_x](https://djitellopy.readthedocs.io/en/latest/tello/#djitellopy.tello.Tello.get_speed_x), [tello.get_speed_y](https://djitellopy.readthedocs.io/en/latest/tello/#djitellopy.tello.Tello.get_speed_y) and [tello.get_speed_z](https://djitellopy.readthedocs.io/en/latest/tello/#djitellopy.tello.Tello.get_speed_z) functions

yorklyb, 2020-11-16:
Thank you so much for your reply. I tried to use pip install https://github.com/damiafuentes/DJITelloPy/archive/master.zip and functions like tello.get_speed_x is available but I feel like they are not working.

I tested the functions on my Apriltag tracking demo, then I got this '[INFO] tello.py -450 - Send command (no expect response) : rc 0 10 0 7', which means I was sending forward velocity and yaw velocity to the drone and it worked. But the return I got from get_speed_x,y,z were all zero. However, I'm pretty sure that the drone had a non-zero velocity because I could see the drone moving forward towards my tag.

yorklyb, 2020-11-16:
> Hey,
>
> yes the `vgx, vgy, vgz` state values are available on regular tellos. Please make sure you have the newest djitellopy installed. The version on pip is outdated. Try installing it using `pip [...]

Thank you so much for your reply. I tried to use pip install https://github.com/damiafuentes/DJITelloPy/archive/master.zip and functions like tello.get_speed_x are available but I feel like they are not working.

I tested the functions on my Apriltag tracking demo, then I got this '[INFO] tello.py -450 - Send command (no expect response) : rc 0 10 0 7', which means I was sending forward velocity and yaw velocity to the drone and it worked. But the return I got from get_speed_x,y,z were all zero. However, I'm pretty sure that the drone had a non-zero velocity because I could see the drone moving forward towards my tag.

M4GNV5, 2020-11-17:
Im not sure wether vgx/vgy/vgz work when using `send_rc_control`. Can you try calling calling `get_speed_x` while performing a regular control command? Something like the following might shine some light on the problem (based on the [record-video.py example](https://github.com/damiafuentes/DJITelloPy/blob/master/examples/record-video.py)):

```python
import time
from threading import Thread
from djitellopy import Tello

tello = Tello()
tello.connect()

keepRunning = True
def statsWriter():
    while keepRunning:
        print("Speeds: ", tello.get_speed_x(), tello.get_speed_y(), tello.get_speed_z())
        time.sleep(0.1)

runner = Thread(target=statsWriter)
runner.start()

printf("Regular controls")
tello.takeoff()
time.sleep(0.5)
tello.move_up(100)
time.sleep(0.5)

print("Switching to RC control")
tello.send_rc_control(50, 0, 0, 0)
time.sleep(0.5)
tello.send_rc_control(-50, 0, 0, 0)
time.sleep(0.5)

tello.land()

keepRecording = False
runner.join()
```

yorklyb, 2020-11-17:
> Im not sure wether vgx/vgy/vgz work when using `send_rc_control`. Can you try calling calling `get_speed_x` while performing a regular control command? Something like the following might shine some [...]

Ok, thanks a lot. I will do it recently.

yorklyb, 2020-11-17:
> Ok, thanks a lot. I will do it recently.

Hi，I tried it just now. But tello.get_speed_x,y,z were still not working (output all zero). I also tried get_accelereation_x,y,z and they worked very well (non-zero output in -0.01g). This is a little bit wried since the drone has zero velocity when it has non-zero acceleration.  I think this is enough for me because I can do the integration by myself to obtain velocity.

closed 2020-11-17 by yorklyb