damiafuentes/DJITelloPy #60: Connection and takeoff error

paulbaumgarten, 2020-11-09:
Hello
Thank you for this amazing library. I have had this working successfully before but am having issues getting it to connect now. It's a new drone, using a new laptop and a fresh install of Python and library.

I've reverted all the way to using code similar to the demo script

```python
    tello = Tello()
    tello.connect()
    time.sleep(1)
    tello.takeoff()
```

The terminal output is as follows

```
tello.py - 186 - Send command: command
tello.py - 203 - Response command: ok
tello.py - 140 - invalid literal for int() with base 10: '-1,-1,-1'
tello.py - 141 - Response was is mid:-1;x:-100;y:-100;z:-100;mpry:-1,-1,-1;pitch:0;roll:0;yaw:0;vgx:0;vgy:0;vgz:0;templ:66;temph:68;tof:6553;h:0;bat:5;baro:118.26;time:0;agx:7.00;agy:0.00;agz:-997.00;

tello.py - 186 - Send command: takeoff
tello.py - 203 - Response takeoff: error
tello.py - 186 - Send command: takeoff
tello.py - 203 - Response takeoff: error
tello.py - 186 - Send command: takeoff
tello.py - 203 - Response takeoff: error
Traceback (most recent call last):
  File "c:/repos/tello-drone-experimenting/demo2-programmed-flight.py", line 44, in <module>
    demo()
  File "c:/repos/tello-drone-experimenting/demo2-programmed-flight.py", line 22, in demo
    tello.takeoff()
  File "c:\repos\tello-drone-experimenting\djitellopy\tello.py", line 334, in takeoff
    if self.send_control_command("takeoff", timeout=20):
  File "c:\repos\tello-drone-experimenting\djitellopy\decorators.py", line 35, in new_f
    return f(*args, **kwds)
  File "c:\repos\tello-drone-experimenting\djitellopy\tello.py", line 273, in send_control_command
    return self.return_error_on_send_command(command, response, self.enable_exceptions)
  File "c:\repos\tello-drone-experimenting\djitellopy\tello.py", line 315, in return_error_on_send_command
    raise Exception(msg)
Exception: Command takeoff was unsuccessful. Message: error
```

I notice that before the exception generated by the takeoff command, there is an "invalid literal" occurring on the connect command. Any advice on what I am missing would be much appreciated.

Regards

M4GNV5, 2020-11-09:
Hey,

please install the latest djitellopy version using `pip install https://github.com/damiafuentes/DJITelloPy/archive/master.zip`. The version on PyPi is from 2018, currently only Damia Fuentes has access to the pypi package but he doesn't work much on djitellopy anymore.

Having `takeoff` return `error` is usually caused by a (nearly) empty battery pack. You can try:

```python
tello = Tello()
tello.connect()
time.sleep(0.5)
print("Battery: ", telo.get_battery())
tello.takeoff()
```

paulbaumgarten, 2020-11-13:
Thank you, that fixed it.

closed 2020-11-13 by paulbaumgarten