damiafuentes/DJITelloPy #63: get_frame_read() frame is None

danpolus, 2020-12-23:
Hi all,
I'm interested getting frames and video from Tello. Already at the first stages, 
after      frame_read = tello.get_frame_read() ,    frame_read.frame  is  None
What might be the problem?
Thanks,
Daniel

This is my code:

import time, cv2
from threading import Thread
from djitellopy import Tello

keepRecording = True

def videoRecorder():
    # create a VideoWrite object, recoring to ./video.avi
    height, width, _ = frame_read.frame.shape
    video = cv2.VideoWriter('video.avi', cv2.VideoWriter_fourcc(*'XVID'), 30, (width, height))

    while keepRecording:
        video.write(frame_read.frame)
        time.sleep(1 / 30)

    video.release()
####################

tello = Tello()
tello.connect()
tello.streamon()

frame_read = tello.get_frame_read()
cv2.imshow('Live Video', frame_read.frame)

recorder = Thread(target=videoRecorder)
recorder.start()

tello.takeoff()
tello.move_up(100)
tello.rotate_counter_clockwise(360)
tello.land()

keepRecording = False

recorder.join() #join thread

M4GNV5, 2020-12-23:
Hey,

your code looks ill formatted. Please use three ``` marks to start or end a code block in github issues.
What is the output of running the program? It looks mostly similar to the record-video example.
Did you try updating your Tello firmware via the official App?

danpolus, 2020-12-24:
Hi,
The firmware version is  01.04.92.01
You are right, this code is similar to the record-video example. But here is a much basic version, that attempts to read a single frame and present it:

```
import time, cv2
from djitellopy import Tello

tello = Tello()
tello.connect()
tello.streamon()

frame_read = tello.get_frame_read()
cv2.imshow('Live Video', frame_read.frame)
```

Here is what I'm getting at the console window:
```
[INFO] tello.py - 423 - Send command: command
[INFO] tello.py - 440 - Response command: ok
[INFO] tello.py - 423 - Send command: streamon
[INFO] tello.py - 440 - Response streamon: ok
Traceback (most recent call last):
  File "C:\Program Files\JetBrains\PyCharm 2019.3.1\plugins\python\helpers\pydev\pydevd.py", line 1434, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "C:\Program Files\JetBrains\PyCharm 2019.3.1\plugins\python\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "D:/My Files/Work/BGU/PhD/Drone BCI/Tello/test.py", line 14, in <module>
    cv2.imshow('Live Video', frame_read.frame)
cv2.error: OpenCV(4.4.0) C:\Users\appveyor\AppData\Local\Temp\1\pip-req-build-6sxsq0tp\opencv\modules\highgui\src\window.cpp:376: error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'cv::imshow'

[INFO] tello.py - 423 - Send command: streamoff
[WARNING] tello.py - 431 - Timeout exceed on command streamoff
[INFO] tello.py - 423 - Send command: streamoff
[WARNING] tello.py - 431 - Timeout exceed on command streamoff
[INFO] tello.py - 423 - Send command: streamoff
[WARNING] tello.py - 431 - Timeout exceed on command streamoff
Exception ignored in: <function Tello.__del__ at 0x000001821E7704C8>
Traceback (most recent call last):
  File "D:\My Files\Work\BGU\PhD\Drone BCI\Tello\venv\lib\site-packages\djitellopy\enforce_types.py", line 45, in wrapper
  File "D:\My Files\Work\BGU\PhD\Drone BCI\Tello\venv\lib\site-packages\djitellopy\tello.py", line 892, in __del__
  File "D:\My Files\Work\BGU\PhD\Drone BCI\Tello\venv\lib\site-packages\djitellopy\enforce_types.py", line 45, in wrapper
  File "D:\My Files\Work\BGU\PhD\Drone BCI\Tello\venv\lib\site-packages\djitellopy\tello.py", line 881, in end
  File "D:\My Files\Work\BGU\PhD\Drone BCI\Tello\venv\lib\site-packages\djitellopy\enforce_types.py", line 45, in wrapper
  File "D:\My Files\Work\BGU\PhD\Drone BCI\Tello\venv\lib\site-packages\djitellopy\tello.py", line 550, in streamoff
  File "D:\My Files\Work\BGU\PhD\Drone BCI\Tello\venv\lib\site-packages\djitellopy\enforce_types.py", line 45, in wrapper
  File "D:\My Files\Work\BGU\PhD\Drone BCI\Tello\venv\lib\site-packages\djitellopy\tello.py", line 466, in send_control_command
  File "D:\My Files\Work\BGU\PhD\Drone BCI\Tello\venv\lib\site-packages\djitellopy\enforce_types.py", line 45, in wrapper
  File "D:\My Files\Work\BGU\PhD\Drone BCI\Tello\venv\lib\site-packages\djitellopy\tello.py", line 512, in raise_result_error
Exception: Command streamoff was unsuccessful. Message: Timeout error!
```

I suspect that get_frame_read() cause a connection drop somehow. This function excecution lasts over 10 sec and returns frame_read.frame = None. During the execution the led goes from blinking green to blinking red. The wifi also disconnects by itselves after some time. 

I must mention, that I don't have any other problems with this API. I successfully fly and control the drone with it's commands.

M4GNV5, 2020-12-25:
The `get_frame_read` function retrieves the first frame in sync. As starting to decode the UDP stream can take some time this is expected to take a few seconds.

For some reason the first frame still seems to be `None`. Running your code example results in the same result for me.
However adding a `time.sleep(1)` between the `tello.get_frame_read` and the `cv2.imshow` fixed the problem.
So i guess the `get_frame_read` does not correctly wait until the OpenCV VideoCapture is initialized.
It works for the take-picture example as the `takeoff` between `get_frame_read` and accesing the image takes some time itself.

Can you verify it works for you with some waiting time or a takeoff between `get_frame_read` and `imshow`? I can check if we can somehow wait for the VideoCapture to be ready.

M4GNV5, 2020-12-26:
After some further testing i got your code example fully working without any changes to the Tello library simply by adding a `cv2.waitKey(100)` at the end, which is required as all UI updates are only executed during `waitKey` calls (see e.g. [this post](https://answers.opencv.org/question/27787/imshow-and-waitkey/))

So the following code works fine for me:

```python
import time, cv2
from djitellopy import Tello

tello = Tello()
tello.connect()
tello.streamon()

frame_read = tello.get_frame_read()
cv2.imshow('Live Video', frame_read.frame)
cv2.waitKey(1000)
```

Walt-H, 2020-12-26:
@M4GNV5 I have replicated the scenario, and that last snippet did the trick 😄

danpolus, 2020-12-29:
I tried inserting time.sleep(1) between get_frame_read and imshow, but it didn't help

Aftrerwards I've updated tello.py with your latest release and now I'm getting an exception during tello.connect():
```
line 529:  raise Exception('Did not receive a state packet from the Tello')
```

M4GNV5, 2020-12-29:
So in fact the OpenCV VideoCapture on your end cannot retrieve frames from the Tello. This is probably some sort of licensing/codec/software issue. Do you have access to a second Tello to verify the issue is on the computer side? 
What python and opencv versions are you running? You said you are using Tello version `01.04.92.01` but for me the newest version is `01.04.91.01`?

danpolus, 2020-12-30:
I succeeded to run and get video using   manual-control-pygame.py   code!
Thanks for the help!

closed 2020-12-30 by danpolus

Walt-H, 2020-12-31:
PyGmae is nice :)

Rishit0210, 2021-12-10:
it is the same problem for me can you tell me how to fix it
i tried this-
[image]
and also this-
[image]

Rishit0210, 2021-12-10:
Cannot see the video feed

Rishit0210, 2021-12-10:
[image]

M4GNV5, 2021-12-10:
@Rishit0210 the problem you are having is not the same as in this issue. Please open a new issue providing your code and the full output (as text).
Also please make sure you are running the latest version of DJITelloPy, preferably even from git, not just the newest release from pypi.

Rishit0210, 2021-12-14:
ok

Rishit0210, 2021-12-14:
@M4GNV5 I have opened an issue please help me out there.

Regards,
Rishit0210

Rishit0210, 2021-12-14:
@M4GNV5 https://github.com/damiafuentes/DJITelloPy/issues/128 
Please help.

Rishit0210, 2021-12-14:
@M4GNV5 https://github.com/damiafuentes/DJITelloPy/issues/128-
 
please help