damiafuentes/DJITelloPy #64: No Video after StreamOn

NerdboyQ, 2021-01-10:
I have been able to successfully connect to the tello and see the camera feed using the app. I have also been able to connect and control the tello over my home network using both a raspberry pi & a macbook pro, however I have not been able to see the camera feed. 

I've tried running the [manual-control-pygame.py](https://github.com/damiafuentes/DJITelloPy/blob/master/examples/manual-control-pygame.py) example, and while the window opens I see no camera feed at all. 

This is the error message I get:`OpenCV: Couldn't read video stream from file "udp://@0.0.0.0:11111"`

M4GNV5, 2021-01-11:
> I have also been able to connect and control the tello over my home network

I assume you are using a Tello EDU in station mode? As noted in the readme:
> When connected to an existing wifi network video streaming is not available.

Sadly this seems to be a limitation of the Tello firmware. At least thats what i recall, the manual doesnt seem to mention it. You could try mailing to ryze support and ask for help.

closed 2021-01-16 by M4GNV5

NerdboyQ, 2021-01-16:
oh wow... that's unfortunate. I did find another library that allowed me to get the video stream for anyone that is interested:

[TelloPy](https://github.com/hanyazou/TelloPy)

M4GNV5, 2021-01-16:
Yes the TelloPy library implements the protocol used by the official app instead of using the API. Afaik the app does not work in station mode at all, but maybe thats a limitation in the app not supporting entering / scanning for an IP adress on a local net?

NerdboyQ, 2021-01-16:
Yes, I connected it to my home router using the ` ap [ssid] [pass] ` and after that I was able to run their video example by changing the default ip address in the library.