damiafuentes/DJITelloPy #65: OpenCV Error Video

Hook12aaa, 2021-01-12:
`tello = Tello()
tello.connect()
tello.streamon()
frame_read = tello.get_frame_read()
tello.takeoff()
tello.move_up(30)
while True:
    img = frame_read.frame
    cv2.imshow("drone", img)
    if cv2.waitKey(1) & 0xff == ord("q"):
        frame_read.stop()
        tello.streamoff()
        tello.land()
        break`

The Error I get:
`
[h264 @ 000001ab71bba300] non-existing PPS 0 referenced
[h264 @ 000001ab71bba300] decode_slice_header error
[h264 @ 000001ab71bba300] no frame!
[h264 @ 000001ab71c17840] error while decoding MB 28 18, bytestream -20`

I can see the video after i get enough errors but it causes problems with flight ( just auto-lands) and it is hard to read the console readout with the many errors.

System is windows 10 , i7 , GeForce GTX

M4GNV5, 2021-01-12:
Hey,

the Tello auto lands after not receiving a command for 15 seconds. In order to prevent this just send it movement commands.

I don't know much about h264, but i assume the errors at start are caused by chunks being received that reference keyframes which were sent before starting to listen to incoming packets. Thus they are somewhat expected.

You can write Tello Logs to a file and examine it independently of the console output if you wish:
```python
import logging
from djitellopy import Tello

handler = logging.FileHandler('tello.log')
handler.setFormatter(Tello.FORMATTER)
Tello.LOGGER.addHandler(handler)

# your program here...
```

Hook12aaa, 2021-01-13:
Is there anyway to get rid of the h264 errors?

M4GNV5, 2021-01-16:
I can't find anything related in the [VideoCapture OpenCV docs](https://docs.opencv.org/3.4.13/d8/dfe/classcv_1_1VideoCapture.html#a8c6d8c2d37505b5ca61ffd4bb54e9a7c). Presumably the output isnt even generated by OpenCV, but ffmpeg or some other decoding library used by OpenCV.
You can try to redirect stdout/stderr. On linux this can be achieved e.g. via `python3 myProgram.py 2>/dev/null` not sure how to do this on windows though. `python3 myProgram.py 2>derp.log` would probably write stderr to a file called derp.log?

TIan1212, 2021-03-07:
There are two issues in your log.
[h264 @ 000001ab71bba300] non-existing PPS 0 referenced
[h264 @ 000001ab71bba300] decode_slice_header error
[h264 @ 000001ab71bba300] no frame!

These lines are saying you didn't receive the reference frame yet. The reference frame is sent periodically every 5 seconds (if I recall correctly). You should need to wait more than 5 seconds to receive the reference frame. After you do, H264 will have sufficient info to decode the stream.

[h264 @ 000001ab71c17840] error while decoding MB 28 18, bytestream -20
Do you leave your wifi password blank on your Tello? if so, your OS might think this network is a public network, and firewall the stream. Just set a password for your Tello. It did the trick for me. Hope it will do the same for you.

TIan1212, 2021-03-07:
Do you leave your wifi password blank on your Tello? if so, your OS might think this network is a public network, and firewall the stream. Just set a password for your Tello. It did the trick for me (MacOS). Hope it will do the same for you.

KBG360, 2021-06-07:
Hello! Any updates on this error? I'm struggling with it as well.

QianMason, 2021-06-17:
I am facing similar issues. Any chance of a resolution?

M4GNV5, 2021-06-17:
What exactly is your problem? You can simply ignore the h264 warning/error outputs.
If your drone auto lands while your program waits for the first frame just wait for the first frame before taking off and/or wait for the first frame in a seperate thread.

Hook12aaa, 2021-07-17:
@M4GNV5  For image processing i need everyframe. UDP doesn't stream all the frames from the camera but instead losses packets. from what i can understand this is a problem with h264