damiafuentes/DJITelloPy #66: name 'client socket' is not defined

BackhaSe, 2021-01-16:
Hello,
My script throws an error when I call me.Connect().

Code:
"""
from djitellopy import Tello
import cv2
import threading
import time

# Image settings
useCamera = 1  # 0 for not using camera and 1 for streaming
width = 640  # width of the image
height = 480  # height of the image

# Flight settings
startCounter = 0  # 0 for testing and 1 for flying

# Thread settings
exitFlag = 0

# Connect to Tello drone
me = Tello()
me.connect()  # This is line 23, error starts here
me.for_back_velocity = 0
me.left_right_velocity = 0
me.up_down_velocity = 0
me.yaw_velocity = 0
me.speed = 0
...
"""

Error:[INFO] tello.py - 424 - Send command: command
Traceback (most recent call last):
  File "C:\***\PycharmProjects\TelloProject\TelloDemo.py", line 23, in <module>
    me.connect()
  File "C:\***\PycharmProjects\DJITelloPy\djitellopy\enforce_types.py", line 45, in wrapper
    return func(*args, **kwargs)
  File "C:\***\PycharmProjects\DJITelloPy\djitellopy\tello.py", line 518, in connect
    self.send_control_command("command")
  File "C:\***\PycharmProjects\DJITelloPy\djitellopy\enforce_types.py", line 45, in wrapper
    return func(*args, **kwargs)
  File "C:\***\PycharmProjects\DJITelloPy\djitellopy\tello.py", line 460, in send_control_command
    response = self.send_command_with_return(command, timeout=timeout)
  File "C:\***\PycharmProjects\DJITelloPy\djitellopy\enforce_types.py", line 45, in wrapper
    return func(*args, **kwargs)
  File "C:\***\PycharmProjects\DJITelloPy\djitellopy\tello.py", line 427, in send_command_with_return
    client_socket.sendto(command.encode('utf-8'), self.address)
NameError: name 'client_socket' is not defined

I'm using PyCharm with Python 3.9 interpreter. As for packages I tried to install the required opencv-python and numpy versions, but I get the error:

ERROR: Could not find a version that satisfies the requirement opencv-python==3.4.3.18
ERROR: No matching distribution found for opencv-python==3.4.3.18

So currently I'm using the closest version I could install (3.4.13.47)

As for my setup I'm using Windows10, i7 and a GTX960. My computer is connected to the internet via LAN and I connect to the drone via WiFi.
I've had this error for quite some time, but while talked to a friend of mine via discord this error didn't appear at all.
Do I have to specify somewhere, which connection or socket PyCharm is supposed to use while executing my code? If yes, where can I do that?
And are the required versions for opencv-python and numpy mandatory or can I use the latest versions as well?

M4GNV5, 2021-01-16:
Hey,

the `client_socket` variable is created by a thread started in the constructor and used in `send_command_with_return` which is used by `command`. I guess this is a race condition where the new thread has not been executed yet and thus the `client_socket` has not been created yet when the call to `me.command()` occurs.

My only guess is that python changed their scheduling behaviour in 3.9 as it works fine with older versions.
I pushed commit f1c1344d9475bb564f33fa2f87d0131cab7894bd (see above) which should fix it. Can you verify this resolves your issue?

BackhaSe, 2021-01-17:
Hello M4GNV5,

I tested yesterday and today a few times and haven't had the client socket - error again. Thank you for this fix.

with kind regard

closed 2021-01-17 by M4GNV5