damiafuentes/DJITelloPy #68: .get_frame_read().frame not producing frames

blairscriven, 2021-01-24:
I am trying to write code to open a window that displays my tello drone's camera view using OpenCV.

I used the following code:

[image]

And it produces the following error:

error: ......\modules\highgui\src\window.cpp:281: error: (-215) size.width>0 && size.height>0 in function cv::imshow

I read in a OpenCV stackoverflow forum that the "size.width>0 && size.height>0" issue suggests  that imshow() is not getting video frames from input-device; meaning my tello drone is not sending image frames back to my computer. I am not sure why this would happen, as the camera on my drone still works and can send video to my smart phone.  Any suggestions? Maybe alternative coding?

M4GNV5, 2021-01-24:
Can you post the full output here? Does it include a `battery?` command? If yes you are using an outdated version of the library. Please make sure you are using the newest version of DJITelloPy, e.g. by installing it from git as noted in the readme:

```
pip install https://github.com/damiafuentes/DJITelloPy/archive/master.zip
```

blairscriven, 2021-01-25:
This is the full error output:

Traceback (most recent call last):
  File "D:/Drone/ImageCapture.py", line 14, in <module>
    cv2.imshow("SpyEye", eye)
cv2.error: OpenCV(4.5.1) C:\Users\appveyor\AppData\Local\Temp\1\pip-req-build-oduouqig\opencv\modules\highgui\src\window.cpp:376: error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'cv::imshow'

I was reading through the API reference, and it stated that "Currently Tello EDUs do not support video streaming while connected to a wifi network". So, I suppose this is why I am not recieving frames?

M4GNV5, 2021-01-25:
please install djitellopy using pip from the git url, not via PyCharm
```
pip install https://github.com/damiafuentes/DJITelloPy/archive/master.zip
```

The version installed via PyCharm is from PyPi and it is very old. I guess you could also fix your issue by adding a `time.sleep(3)` between `get_frame_read` and `imshow` like shown in the following code. This is because you are using a very old version of the library which does not wait for the first frame to be received before returning from `tello.get_frame_read` 

```python
#...
drone.streamon()
frame_read = tello.get_frame_read()

time.sleep(3)

while True:
    cv2.imshow("SpyEye", frame_read.frame)
    if cv2.waitKey(1) & 0xff == ord('q'):
        break
# ...
```

> I was reading through the API reference, and it stated that "Currently Tello EDUs do not support video streaming while connected to a wifi network". So, I suppose this is why I am not recieving frames?

This refers to a Tello EDU drone which is connected to an existing WiFi network instead of creating its own WiFi hotspot (after calling `tello.connect_to_wifi(ssid, password)`). Is this the case with your setup?

UltraBob, 2021-01-25:
This is probably a naive question, but why not update the version on Pypi so that people get the updated code?

M4GNV5, 2021-01-25:
Currently only Damia (the owner of this repo) has access to the pypi package. He does not spend much time on github, which is why he gave me maintainer permissions for this repo, but i currently do not have access to the pypi package.

I guess i could create a `djitellopy2` package :thinking:

M4GNV5, 2021-01-25:
Ok you can now install [the djitellopy2 package](https://pypi.org/project/djitellopy2/) either via pip, via the PyCharm IDE, etc.
I assume this closes this issue?

closed 2021-01-25 by M4GNV5