damiafuentes/DJITelloPy #70: AttributeError: 'Tello' object has no attribute 'background_frame_read'

andrew-elmore, 2021-01-27:
Hi All, I am trying out the take-picture.py example and am getting the following error, I have updated firmware, tried adding a sleep(3) after streamon(), and am not using an EDU.

```
[ERROR] tello.py - 83 - derp
[INFO] tello.py - 420 - Send command: command
[INFO] tello.py - 437 - Response command: ok
[INFO] tello.py - 420 - Send command: streamon
[INFO] tello.py - 437 - Response streamon: ok
Traceback (most recent call last):
  File "/Users/andrewelmore/PycharmProjects/drone2/main.py", line 8, in <module>
    frame_read = tello.get_frame_read()
  File "/Users/andrewelmore/PycharmProjects/drone2/venv/lib/python3.9/site-packages/djitellopy/tello.py", line 401, in get_frame_read
    if self.background_frame_read is None:
AttributeError: 'Tello' object has no attribute 'background_frame_read'
[INFO] tello.py - 420 - Send command: streamoff
[WARNING] tello.py - 428 - Timeout exceed on command streamoff
[INFO] tello.py - 420 - Send command: streamoff
```

Any help would be appreciated.

M4GNV5, 2021-01-27:
Where does the log line
`[ERROR] tello.py - 83 - derp` come from? Did you modify the `tello.py` file?

The `background_frame_read` is [declared and initialized here](https://github.com/damiafuentes/DJITelloPy/blob/master/djitellopy/tello.py#L70), did you change something in there? Otherwise it could be a new behaviour in python 3.9, but that would be weird.

andrew-elmore, 2021-01-27:
that fixed it, the 

```
Optional['BackgroundFrameRead'] = None
```

was missing, no idea how I messed it up but thank you for the help

closed 2021-01-27 by andrew-elmore

blairscriven, 2021-01-27:
I actually had the same issue when I installed djitellopy2. I looked at the tello.py file I had and that section of the code looks like this:

[image]

M4GNV5, 2021-01-27:
Indeed, for whatever reason the code in djitellopy2 was outdated. Im new to python packaging, but i can't explain how this happened.

Anyways i released version [2.2 of djitellopy2](https://pypi.org/project/djitellopy2/2.2) now with up to date code.

blairscriven, 2021-01-28:
Thanks for all the quick and valuable help you've been, M4GNV5!