damiafuentes/DJITelloPy #71: Exception: Did not receive a state packet from the Tello

radik86, 2021-01-29:
Hello,
I wanted to start working with djitellopy lib. When I use the example code I receive the below mentioned error.
The error appears in the command takeoff(). I see that the initial command "command" was send and a receive a response from tello. But  after connect() the takeoff() causing the error. Have you any ideas? 
I am able to control my tello ryze with the original SDK. Thank you in advance.

 File "c:\users\...\djitellopy\djitellopy\enforce_types.py", line 49, in wrapper
    return func(*args, **kwargs)
  File "c:\users\...\djitellopy\djitellopy\tello.py", line 526, in connect
    raise Exception('Did not receive a state packet from the Tello')
Exception: Did not receive a state packet from the Tello

M4GNV5, 2021-01-29:
Hey,

The error comes from [here](https://github.com/damiafuentes/DJITelloPy/blob/master/djitellopy/tello.py#L526). After a succesfull connection with the tello (i.e. receiving an `ok` response to the `command` command) the lib waits for the first state packet to arrive before continuing.

Can you paste the full output?
What Tello firmware are you using? Can you try updating it using the official App?
You can also call `tello.connect(False)` which disables the witing for a state packet. Does the `takeoff` command afterwards works?

Jheronymus, 2021-01-29:
Hi, I have the same problem. I can fly the Tello drone (not EDU) normally with the standard app. But won't connect with library.
App version: 1.6.0.0
Firmware version: 01.04.92.01 (The app reports there is an update, but that one applies to EDU model, in the text it mentioned a lower version for the standard model, quite weird)
Loader version: 00.01.00.00

I'm running Python 3.7.9 on windows 10 via Visual Studio Code. I installed with `pip3 install djitellopy2`
Output With Connect()
`[INFO] tello.py - 104 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 418 - Send command: 'command'
[INFO] tello.py - 443 - Response command: 'ok'
Traceback (most recent call last):
  File "c:../Documents/DroneProgram/test.py", line 17, in <module>
    me.connect()
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\enforce_types.py", line 49, in wrapper
    return func(*args, **kwargs)
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\tello.py", line 526, in connect
    raise Exception('Did not receive a state packet from the Tello')
Exception: Did not receive a state packet from the Tello`

output with Connect(False)
`[INFO] tello.py - 104 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 418 - Send command: 'command'
Traceback (most recent call last):
  File "c:../Documents/DroneProgram/test.py", line 17, in <module>
    me.connect(False)
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\enforce_types.py", line 49, in wrapper
    return func(*args, **kwargs)
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\tello.py", line 514, in connect
    self.send_control_command("command")
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\enforce_types.py", line 49, in wrapper
    return func(*args, **kwargs)
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\tello.py", line 462, in send_control_command
    response = self.send_command_with_return(command, timeout=timeout)
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\enforce_types.py", line 49, in wrapper
    return func(*args, **kwargs)
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\tello.py", line 441, in send_command_with_return
    response = response_decoded.rstrip("\r\n")
UnboundLocalError: local variable 'response_decoded' referenced before assignment`

output with Connect(True) <= same as Connect()
`[INFO] tello.py - 104 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 418 - Send command: 'command'
[INFO] tello.py - 443 - Response command: 'ok'
Traceback (most recent call last):
  File "c:../Documents/DroneProgram/test.py", line 17, in <module>
    me.connect(True)
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\enforce_types.py", line 49, in wrapper
    return func(*args, **kwargs)
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\tello.py", line 526, in connect
    raise Exception('Did not receive a state packet from the Tello')
Exception: Did not receive a state packet from the Tello`

M4GNV5, 2021-01-29:
Hey,

thanks for the output. Can you add the following to the start of your file:
```python
import logging
Tello.LOGGER.set_level(logging.DEBUG)
```

This makes logging more verbose, and should enable logging when a state packet is received, this way we can find out wether the problem is with passing the state to the tello instance or with receiving state packets alltogether.

Jheronymus, 2021-01-29:
Hi,

tried to do that from my script, but that didn't work. AttributeError: 'Logger' object has no attribute 'set_level'
Had to peak into the actual tello.py file and changed line 47 to DEBUG

Results in:
```
[INFO] tello.py - 104 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[DEBUG] tello.py - 415 - Waiting 0.0019943714141845703 seconds to execute command: command...
[INFO] tello.py - 418 - Send command: 'command'
[DEBUG] tello.py - 123 - Data received from 192.168.10.1 at client_socket
[DEBUG] tello.py - 123 - Data received from 192.168.10.1 at client_socket
Traceback (most recent call last):
  File "c:../Documents/GaliaDroneChallenge/test.py", line 21, in <module>
    me.connect()
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\enforce_types.py", line 49, in wrapper
    return func(*args, **kwargs)
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\tello.py", line 514, in connect
    self.send_control_command("command")
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\enforce_types.py", line 49, in wrapper
    return func(*args, **kwargs)
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\tello.py", line 462, in send_control_command
    response = self.send_command_with_return(command, timeout=timeout)
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\enforce_types.py", line 49, in wrapper
    return func(*args, **kwargs)
  File "C:..\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\tello.py", line 441, in send_command_with_return
    response = response_decoded.rstrip("\r\n")
UnboundLocalError: local variable 'response_decoded' referenced before assignment
```

M4GNV5, 2021-01-29:
Does this happen multiple times or just once?
It looks like you receive some non utf-8 bytes from the tello, which makes the try/except [here](https://github.com/damiafuentes/DJITelloPy/blob/master/djitellopy/tello.py#L435-L441) fail. Does trying it a second time change anything?
Also it's `logging.DEBUG` (from the `import logging` not the `Logger` instance from `Tello.LOGGER`)

Jheronymus, 2021-01-30:
hmm... weird indeed. The logging I posted above only happens on first try after I connected to the wifi. On second and further attempts it does goes past the OK receive
But the problem is that it just wont accept the state message. It does not read from the state port. Curious, because this has been implemented slightly different to the client_socket

Before it raises the exception 'Did not receive state packet from Tello'
It just loops over the 20 REPS in the connect method. Added logging in get_own_udp_object which prints 20 times:
[DEBUG] tello.py - 110 - {'192.168.10.1': {'responses': [], 'state': {}}}

Jheronymus, 2021-01-30:
I tried your commit and now the output is consistant with the simple test script. Still no logging on the state...
```
[INFO] tello.py - 104 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[DEBUG] tello.py - 415 - Waiting 0.0 seconds to execute command: command...
[INFO] tello.py - 418 - Send command: 'command'
[DEBUG] tello.py - 123 - Data received from 192.168.10.1 at client_socket
[INFO] tello.py - 442 - Response command: 'ok'
Traceback (most recent call last):
  File "c:/Users/.../Documents/GaliaDroneChallenge/test2.py", line 7, in <module>
    tello.connect()
  File "C:\Users\...\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\enforce_types.py", line 49, in wrapper
    return func(*args, **kwargs)
  File "C:\Users\...\AppData\Local\Programs\Python\Python37\lib\site-packages\djitellopy\tello.py", line 524, in connect
    raise Exception('Did not receive a state packet from the Tello')
Exception: Did not receive a state packet from the Tello
```

radik86, 2021-01-30:
This is my current output, also when i using the logging.

PS C:\Users\user\source\iotrepos\>  & 'C:\Users\user\AppData\Local\Programs\Python\Python38-32\python.exe' 'c:\Users\user\.vscode\extensions\ms-python.python-2021.1.502429796\pythonFiles\lib\python\debugpy\launcher' '61392' '--' 'c:\Users\user\source\iotrepos\tello-command.py'
[INFO] tello.py - 104 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[DEBUG] tello.py - 415 - Waiting 0.00973367691040039 seconds to execute command: command...
[INFO] tello.py - 418 - Send command: 'command'
[DEBUG] tello.py - 123 - Data received from 192.168.10.1 at client_socket
[INFO] tello.py - 443 - Response command: 'ok'
Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\Users\user\.vscode\extensions\ms-python.python-2021.1.502429796\pythonFiles\lib\python\debugpy\__main__.py", line 45, in <module>
    cli.main()
  File "c:\Users\user\.vscode\extensions\ms-python.python-2021.1.502429796\pythonFiles\lib\python\debugpy/..\debugpy\server\cli.py", line 444, in main
    run()
  File "c:\Users\user\.vscode\extensions\ms-python.python-2021.1.502429796\pythonFiles\lib\python\debugpy/..\debugpy\server\cli.py", line 285, in run_file
    runpy.run_path(target_as_str, run_name=compat.force_str("__main__"))
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\runpy.py", line 265, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\runpy.py", line 97, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\Users\user\source\iotrepos\tello-command.py", line 9, in <module>
    me.connect()
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\site-packages\djitellopy\enforce_types.py", line 49, in wrapper      
    return func(*args, **kwargs)
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\site-packages\djitellopy\tello.py", line 526, in connect
    raise Exception('Did not receive a state packet from the Tello')
Exception: Did not receive a state packet from the Tello

When i install the pjitellopy instead of pjitellopy2  lib and use it I am able to takeoff and get battery state, but after that receive only timeout issues:
my Code:
from djitellopy import Tello
import logging
#Tello.LOGGER.setLevel(logging.DEBUG)

me = Tello()
me.connect()
#me.connect(False)
me.get_battery()

me.takeoff()
me.move_left(30)
me.land()

Console log:
Send command: command
Response: b'ok'
Send command: battery?
Response: b'82\r\n'
Send command: takeoff
Timeout exceed on command takeoff
Command takeoff was unsuccessful. Message: False
Send command: left 30
Timeout exceed on command left 30
Command left 30 was unsuccessful. Message: False
Send command: land
Response: b'error Not joystick'
Command land was unsuccessful. Message: error Not joystick

Using the the original SDK is working fine for me. I first assumed, that some how my surface is blocking the connection.

Thank you for your support, really appreciate!

M4GNV5, 2021-01-30:
@radik86 the takeoff command can take some time. In the new version there is a larger delay waiting for it. Takeoff time can be reduced by having better light conditions when flying (turn on the light, dont fly at night).

@Jheronymus yes, it looks like the first packet received contains some weird data. Not sure what it is, it might be interesting to print it or inspecting it with wireshark and check what it contains, but it does not really matter.

My only guess left is that an antivirus or (windows-)firewall blocks python from receiving the UDP packets. If you need state information you can look into that / try disabling them etc.. I don't use Windows and can't reproduce the issue.
Alternatively if you dont care about state packets you can try something like the following:

```python
import time
from djitellopy import Tello

tello = Tello()

tello.connect(False)
time.sleep(1)
tello.takeoff()

# more commands here...

tello.land()

```

Jheronymus, 2021-01-31:
Hi Jakob, indeed. My firewall was the problem. Some other application made a hole for the control port, but I had to add the other two ports manually. Never occurred to me, because messages were showing up in wireshark.

- 8889/udp  control
- 8890/udp state
- 11111/udp video

Thanks for the support.

radik86, 2021-02-01:
I also can confirm, that my issue was caused by blocked inbound connections for the udp ports.

closed 2021-02-01 by M4GNV5

mrsojourn, 2021-02-08:
@M4GNV5 @radik86 @Jheronymus Hi guys!
I'm having the same issue, 
i did try to add all these udp ports manually via inbound rules (firewall advanced security) but i'm still getting the same error.
"Did not receive a state packet from the Tello"
please correct me i did smth wrong or can you explain how you did it?

M4GNV5, 2021-02-14:
Did you try temporarily disabling your firewall? Are you running any sort of antivirus software?

Alfredo-Nantes, 2021-02-16:
> @M4GNV5 @radik86 @Jheronymus Hi guys!
> I'm having the same issue,
> i did try to add all these udp ports manually via inbound rules (firewall advanced security) but i'm still getting the same error. [...]

Same here. Running on Mac Catalina, Py3, no firewall same error message "Did not receive a state packet from the Tello".

I see that Tello receives commands (e.g."command", "streamon") and sends responses back to port 8890 (control) or 11111 (video streaming), I've checked with a packet sniffer (Wireshark).
But then the client is not able to handle them, as if state_socket.recvfrom() or cv2.VideoCapture(address) could not see the stream and you get either "Did not receive a state packet from the Tello" or "OpenCV: Couldn't read video stream from file "udp://@0.0.0.0:11111"".
This problem only appears on my Mac, but on my Windows machine it all works fine :-(

mrsojourn, 2021-02-16:
@Alfredo-Nantes hi guys, 
yes @M4GNV5 :), Disabling my firewall did help me solve the issue!!
but the issue is im using my university's laptop and i dont have a full control over my laptop's administrator, 
if i could fix it through the udp port setting that would be awesome !!

Alfredo-Nantes, 2021-02-16:
@mrsojourn  and @M4GNV5 I confirm PF was the problem for me on my MAc.
@mrsojourn maybe you can edited /etc/pf.conf and add lines like these:
pass in proto udp from any to any port 11111
pass in proto udp from any to any port 8889
pass in proto udp from any to any port 8890
or whatever ports you have defined in your tello.py
You'll still need to disable and re-enable PF after that (sudo pfctl -d; sudo pfctl -E) so you would still need admin access.

mrsojourn, 2021-02-16:
@Alfredo-Nantes Cheers mate! you made my day :)
@M4GNV5 Thankyou for your help :)

tpt5cu, 2021-04-26:
I am now encountering this issue on my Linux HyperV VM

LuanBaviloni, 2021-06-13:
Editing the PF config on MAC wasn't enough for me.

The problem was solved after I updated the Tello firmware.

milaomrani, 2023-08-23:
It wasn't solved until I changed my IDE from VScode to Spyder.

SametGul, 2024-07-02:
I had the same problem on my Ubuntu 22.04 PC, after closing firewall, the problem solved.