damiafuentes/DJITelloPy #73: error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'cv::imshow'

mrsojourn, 2021-02-07:
I'm having
 error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'cv::imshow'
while i try to get a video feed from my tello drone.
Can someone please help me fix it?

from djitellopy import Tello
import cv2
from time import sleep

me = Tello()
me.connect()
print(me.get_battery())
me.streamoff()
me.streamon()

while True:
    img = me.get_frame_read().frame
    cv2.imshow('Image', img)
    cv2.waitKey(1)
[image]

M4GNV5, 2021-02-07:
Can you provide a full screenshot of the output? the `print(me.get_battery())` part seems to be missing in the output.

I guess (can't be sure without the full screenshot) you are using an old version of DJITelloPy. Please try running:
```bash
pip uninstall djitellopy
pip install djitellopy2
```

mrsojourn, 2021-02-08:
Hi @M4GNV5, thanks for your reply!
I've installed djitellopy2 . Now i'm running into an another issue which says "Did not receive a state packet from the Tello"

Another thing which i've noticed with djitellopy (before i uninstall it) was my drone was able to takeoff but the output terminal displays returns takeoff command was unsuccessful!!

[image]

M4GNV5, 2021-02-08:
It sounds like your drone receives your packets, but you dont receive its state packets back. This is exactly [the same as in this issue](https://github.com/damiafuentes/DJITelloPy/issues/71#issuecomment-770442218).
please make sure your firewall allows the following ports:
- 8889/udp control
- 8890/udp state
- 11111/udp video

closed 2021-02-28 by M4GNV5