damiafuentes/DJITelloPy #74: can't install opencv-python version 3.4.3.18

samlaf, 2021-02-10:
When I run
`pip3 install -r requirements.txt`
I get the following error:

> ERROR: Could not find a version that satisfies the requirement opencv-python==3.4.3.18 (from versions: 3.4.8.29, 3.4.9.31, 3.4.9.33, 3.4.10.35, 3.4.10.37, 3.4.11.39, 3.4.11.41, 3.4.11.43, 3.4.11.45, 3.4.13.47, 4.1.2.30, 4.2.0.32, 4.2.0.34, 4.3.0.36, 4.3.0.38, 4.4.0.40, 4.4.0.42, 4.4.0.44, 4.4.0.46, 4.5.1.48)
> ERROR: No matching distribution found for opencv-python==3.4.3.18
>

samlaf, 2021-02-10:
Update: I installed following readme with `pip3 install -e .` and this installed

> numpy-1.20.1
> opencv-python==3.4.3.18 

which I'm guessing are the latest versions.
I suggest either updating the requirements.txt file (and the readme, to suggest installing via `pip3 install -r requirements.txt -e .`) or just getting rid of it altogether (with risk of new versions of numpy and opencv breaking some bode here)

M4GNV5, 2021-02-10:
I'm confused. You said:
> ERROR: Could not find a version that satisfies the requirement opencv-python==3.4.3.18

but then:

> I installed following readme with `pip3 install -e .` and this installed
> opencv-python==3.4.3.18

:confused:

samlaf, 2021-02-11:
My bad! I meant that `pip install -e .` installed opencv-python version 4.5.1.48 (most recent I'm guessing).

closed 2021-02-14 by M4GNV5

lkup-hash, 2022-07-08:
I am also receiving the same error as the original poster, however, I have not been able to fix the issue. Any ideas on whats going wrong?

Sidkotian2001, 2022-08-07:
Hey i was facing the same error, but was able to solve it by installing opencv-python==3.4.3.18 using pip2

BhavikaC09, 2023-02-05:
@Sidkotian2001 can you provide the solution in a detailed manner? I've been trying to fix it since the past few days and still haven't been able to solve it

Sidkotian2001, 2023-02-06:
So what I think is OpenCV version 3.4.3.18 is not made for python3 but for python2. Hence it shows that specific error message on trying to install using pip3.

However I tried installing it using python pip2 package manager using the exact same way, it worked that is : 

pip2 install python-opencv==3.4.3.18

Hope this helps

commit 9c3f61bc (2021-02-14): bump required numpy and opencv versions (fixes #74)