damiafuentes/DJITelloPy #75: Example mutithreading for videostream?

Matthias84, 2021-02-14:
Hi, I was wondering in how we can use the lib to control the drone and receive video stream in parallel?
Internally the frame reader is already a separate thread, but I'm not sure how you can embedd it in a mainloop based framewok as paygame for example?

M4GNV5, 2021-02-14:
If you want a continuous video stream you have to create a separate thread yourself. [The record-video example](https://github.com/damiafuentes/DJITelloPy/blob/master/examples/record-video.py) shows how this can be done.

For pygame you could then for example receive frames and user input in the main thread, send the user input to a separate thread and call tello movement functions (such as e.g. `move_forward`) in the second thread.

The following code is a very basic example of what i mean, it is a combination of [the record-video](https://github.com/damiafuentes/DJITelloPy/blob/master/examples/record-video.py) and [the pygame](https://github.com/damiafuentes/DJITelloPy/blob/master/examples/manual-control-pygame.py) examples.
It might be a good idea to use [a mutex](https://docs.python.org/3/library/threading.html#lock-objects) or even [a queue](https://docs.python.org/3/library/queue.html), i just tried to keep the below code simple.

```python
import time
import pygame
from djitellopy import Tello
from threading import Thread

pygame.init()
tello = Tello()
lastKey = None

# this function running in a seperate thread will call the tello movement functions
def movementWorker():
    global lastKey
    while True:
        if lastKey == pygame.K_w:
            tello.move_forward(30)
        else:
            time.sleep(0.01)
        # TODO more movement commands...
        lastKey = None

# create and start the movement thread
Thread(target=movementWorker).start()

# timer to update the frame
screen = pygame.display.set_mode([960, 720])
pygame.time.set_timer(pygame.USEREVENT + 1, 1000 // 30)

tello.streamon()
frame_read = self.tello.get_frame_read()
tello.takeoff()

for event in pygame.event.get():
    if event.type == pygame.QUIT:
        exit(0)
    elif event.type == pygame.KEYDOWN:
        lastKey = event.key
    elif event.type == pygame.USEREVENT + 1:
         frame = pygame.surfarray.make_surface(frame_read.frame)
         screen.blit(frame, (0, 0))
         pygame.display.update()
```

closed 2021-02-28 by M4GNV5

Matthias84, 2021-03-01:
Sorry that I didn't left a reply yet. Thank you very much for your snippet, but as a first step I started to work on a pygame cockpit using the tellopy lib.  It also offers a pretty pythonic way to deal with events, ubt lacks SDK2.0 commands. I will see how much I can realize and if I learn more about good tello control coding :+1: