damiafuentes/DJITelloPy #76: no valid imu after multiple calibrations

brandonsin197, 2021-02-19:
hi, I am having issues using the drone commands such as drone; rotate, left, right etc.. every time I try to run one of these commands I get the error no valid imu shortly followed by a crash. I have looked at the suggestions for this issue and I have calibrated and updated the drone multiple times to no avail, there is also adequate lighting as well. I can however run commands manually using drone.send_rc_control which runs without issue so I am using that for now but would like to switch to the commands for ease of use, any suggestions would be great, thanks. 

python 3.7

msfisher, 2021-12-01:
I am experiencing the same error

hamzagorgulu, 2023-09-15:
I am having the same issue, anyone solved the problem?