damiafuentes/DJITelloPy #77: Latency Question

samlaf, 2021-02-19:
Are people experiencing a 2s latency when using this protocol/library?

When using the (unofficial) low-level protocol from the [TelloPy](https://github.com/hanyazou/TelloPy) repo, I get a very very low (<50ms) latency, which makes the driving around experience much more enjoyable than when I'm using this library.

Any reason as to why?

M4GNV5, 2021-02-28:
I also have noticable latencies, but i wouldnt say they are ~2s.
There is [this stackoverflow thread](https://stackoverflow.com/questions/30032063/opencv-videocapture-lag-due-to-the-capture-buffer) which you could try, i.e. reducing the buffer of the VideoCapture object.

For a quick test adding this line to one of the examples should alrea:
```python
frame_read.cap.set(cv2.CAP_PROP_BUFFERSIZE, 2)
```

If that doesnt help i can only assume the Tello buffers internally, which we cant really affect.

samlaf, 2021-02-28:
I rechecked and you're right, its probably 300-800ms latency. Not that bad, but makes it impossible to use for fpv flying.

I tried setting the CAP_PROP_BUFFERSIZE property and I just get 'False' as return. `dc1394` is installed on my machine, yet `opencv` doesn't seem to recognize it. Nonetheless, `dc1394` is a camera library, which I'm guessing would then only affect camera capturing + streaming (so what the Tello is doing on board), and not the camera reading which we are doing (which uses the FFMPEG backend).

M4GNV5, 2021-04-21:
I'm closing this in favor of #87, even though this issue is older the new one has more activity.

closed 2021-04-21 by M4GNV5