damiafuentes/DJITelloPy #78: Cannot import 'TelloSwarm'

MichalKormanik, 2021-02-24:
Hello, I'm new with this, i tried to swarn two Tello EDUs, connected them to same WiFi, but when i try to run your example code it displays this error: 
`Traceback (most recent call last):
    File "/Users/michal/PycharmProjects/TelloEdu/venv/swarmTello.py", line 1, in <module>
      from djitellopy import TelloSwarm
ImportError: cannot import name 'TelloSwarm' from 'djitellopy' (/Users/michal/PycharmProjects/TelloEdu/venv/lib/python3.7/site-packages/djitellopy/__init__.py)
Process finished with exit code 1`

Basic control with video stream works perfectly, but i can't get swarm running.
Can you help me resolve this issue?

M4GNV5, 2021-02-24:
Please try installing the `djitellopy2` packet. I assume you currently have the old version installed.
Try:
```
pip3 install djitellopy2
```

MichalKormanik, 2021-02-24:
I have installed djitellopy 1.5 and djitellopy2 2.2.

MichalKormanik, 2021-02-24:
I reinstalled it and error disappeared but new one poped up:

> [INFO] tello.py - 104 - Tello instance was initialized. Host: '10.0.0.125'. Port: '8889'.
[INFO] tello.py - 104 - Tello instance was initialized. Host: '10.0.0.126'. Port: '8889'.
Traceback (most recent call last):
  File "/Users/michal/PycharmProjects/telloEdu/venv/swarmTello.py", line 6, in <module>
    "10.0.0.126"
  File "/Users/michal/PycharmProjects/TelloEdu/venv/lib/python3.7/site-packages/djitellopy/enforce_types.py", line 49, in wrapper
    return func(*args, **kwargs)
  File "/Users/michal/PycharmProjects/TelloEdu/venv/lib/python3.7/site-packages/djitellopy/swarm.py", line 46, in fromIps
    return TelloSwarm(tellos)
  File "/Users/michal/PycharmProjects/TelloEdu/venv/lib/python3.7/site-packages/djitellopy/enforce_types.py", line 48, in wrapper
    check_types(spec, *args, **kwargs)
  File "/Users/michal/PycharmProjects/TelloEdu/venv/lib/python3.7/site-packages/djitellopy/enforce_types.py", line 41, in check_types
    raise TypeError('Unexpected type for \'{}\' (expected {} but found {})'.format(name, type_hint, type(value)))
TypeError: Unexpected type for 'tellos' (expected typing.List[djitellopy.tello.Tello] but found <class 'list'>)

Process finished with exit code 1

M4GNV5, 2021-02-24:
Sharing your code would make it easier to help you.

But i assume you have something like this in your code:
```python
ips = [
    "10.0.0.126",
    "10.0.0.127",
    # ...
]
swarm = TelloSwarm(ips)
```

The last line is incorrect, the constructor requires a list of `Tello` instances. If you have a list of IPs you can use the [TelloSwarm.fromIps](https://djitellopy.readthedocs.io/en/latest/swarm/#djitellopy.swarm.TelloSwarm.fromIps) function.:
```python
swarm = TelloSwarm.fromIps(ips)
```

MichalKormanik, 2021-02-24:
Sorry i forgot to mentioned it. I just edited your Example code. Replaced with my IPs and commented out actions that i dont want them to do. So the will just take off and land, here is whole code:
```
from djitellopy import TelloSwarm

swarm = TelloSwarm.fromIps([
    "10.0.0.125",
    "10.0.0.126"
])

swarm.connect()
swarm.takeoff()

# run in parallel on all tellos
#swarm.move_up(100)

# run by one tello after the other
#swarm.sequential(lambda i, tello: tello.move_forward(i * 20))

# making each tello do something unique in parallel
#swarm.parallel(lambda i, tello: tello.move_left(i * 100))

swarm.land()

```

closed 2021-02-28 by M4GNV5

M4GNV5, 2021-02-28:
Hey,

it looks like the type checks for complex types such as the `List[Tello]` type used in `TelloSwarm` were actually broken. I hopefully fixed it in the above commit.
Please run
```
pip install djitellopy2 --upgrade
```

MichalKormanik, 2021-03-01:
I just tried it. I works very well!
Thank you!

commit 6235f605 (2021-02-28): fix type checks for complex types such as List[int] (fixed #78)