damiafuentes/DJITelloPy #79: Error

Hook12aaa, 2021-03-01:
I honestly don't know what I did differently this entire setup worked last time.
Dependencies:

-  I have a lan cable and a wifi card. The wifi card is connected to the tello drone
-  DJITelloPy 
- CV2

This was just a hack to attempt to connect the drone.
`
from DJITelloPy import Tello
import cv2.cv2 as cv2
drone = Tello()
drone.connect()

def screen_shot(drone):
   drone.streamon()
   frame_read = drone.get_frame_read()
   for i in range(100):
      img = frame_read.frame
      grey = cv2.cvtColor(img,cv2.COLOR_BGR2GRAY)
   frame_read.stop()
   return img

cv2.imwrite("../capture.jpeg",screen_shot(drone))
`
The error I got and yes the drone was connected to the computer. I wish I knew why
`
[INFO] tello.py - 424 - Send command: command
[INFO] tello.py - 441 - Response command: ok
Traceback (most recent call last):
  File "C:\Users\anton\Documents\cross-colab\drone-project\OpenCV-drones\test.py", line 4, in <module>
    drone.connect()
  File "C:\Users\anton\Documents\cross-colab\drone-project\OpenCV-drones\DJITelloPy\enforce_types.py", line 45, in wrapper
    return func(*args, **kwargs)
  File "C:\Users\anton\Documents\cross-colab\drone-project\OpenCV-drones\DJITelloPy\tello.py", line 529, in connect
    raise Exception('Did not receive a state packet from the Tello')
Exception: Did not receive a state packet from the Tello
`

Hook12aaa, 2021-03-02:
Disable your firewall for the flight duration.
To do this on PC:
-Go to Control Panel
-Go to System and Security
-Go to windows defender  and disable it 

** Be warned that if you disable the private windows defender your drone may be on public **

For mac:
edited /etc/pf.conf and add lines like these:
pass in proto udp from any to any port 11111
pass in proto udp from any to any port 8889
pass in proto udp from any to any port 8890
You'll still need to disable and re-enable PF after that (sudo pfctl -d; sudo pfctl -E) so you would still need admin access.

^^ as mentioned by someone on ticket 71
This all comes from ticket 71

closed 2021-04-21 by M4GNV5

tpt5cu, 2021-04-26:
What are the steps to fix if running on a VM? Tried disabling firewall. Did not work.

msfisher, 2021-11-30:
turning off the firewall did not work for me on windows 10, using python 3.7 ... I still get the exception "Did not receive a state packet from the Tello"