damiafuentes/DJITelloPy #83: something strange

djhnsn, 2021-03-13:
I have some python programs on my jetson nano and laptop to fly this thing. They used to work just fine. I have just tried them in 2 environments, on my jetson nano ubuntu 18.04 and on my laptop ubuntu 20.04 . In both cases when I send a takeoff command, it takes off, flies to the right, lands, takes off, lands and shuts off. 

I thought maybe something was wrong with the drone, but I tried it on the tello app and it flies normally. 

Has anyone experienced something like this?

M4GNV5, 2021-03-13:
Please try calibrating your drone via the settings menu in the app.

djhnsn, 2021-03-13:
Thanks. I just tried it, now it does act a little differently. It flies straight up and shuts off. Maybe it is a problem with the drone. Something I noticed is that the wifi signal meter on both computers looks like it's dropping connection to the tello. 

I have a spare tello, I will dig it out in a little while and see if it works correctly.

djhnsn, 2021-03-13:
Yeah a second tello worked just fine. I am not sure why the first one works well in the app and not from my computers. I guess I need to research this problem a little bit. 

Anyway,
thanks for the help.

closed 2021-03-13 by djhnsn