damiafuentes/DJITelloPy #87: OpenCV Video Feed Latency

Fishoupolis, 2021-04-14:
Does anyone know how to reduce video feed latency? Everything is fine except for the video feed, which has about a 1-second latency. On the Tello mobile app from Ryze, the video latency is unnoticeable. However, the video latency when using this library is very noticeable.

G-Rizzle, 2021-04-15:
for me, i cant see any latency. try
import cv2

drone = tello.Tello()
drone.connect()

while True:
    img = drone.get_frame_read().frame
    cv2.imshow('stream', img)
    cv2.waitKey(1)

took it from here, worked fine for me
https://www.youtube.com/watch?v=LmEcyQnfpDA&t=12027s

Fishoupolis, 2021-04-15:
Hmm... I tried it but the latency is still there.

Fishoupolis, 2021-04-15:
https://user-images.githubusercontent.com/82547919/114942460-6ca37000-9df9-11eb-801d-d4617ccd366e.mov

This is what the latency looks like

G-Rizzle, 2021-04-16:
hmm..no, i dont have that. maybe your computer is the problem?
 

Gesendet: Donnerstag, 15. April 2021 um 23:47 Uhr

 

https://user-images.githubusercontent.com/82547919/114942460-6ca37000-9df9-11eb-801d-d4617ccd366e.mov

Fishoupolis, 2021-04-16:
My computer specs are :

Ryzen 5 3600
RTX 2070 
16GB DDR4
Windows 10 64bit

What are your pc specs?

Fishoupolis, 2021-04-17:
Ok, I'm pretty sure that I have this latency due to my IDE, which is Pycharm. Pycharm doesn't support GPU processing and therefore cannot process the video feed as quickly. What IDE are you using?

RalphBariz, 2021-04-18:
Can confirm the lag (about a sec) for me too.
i7-7820X
32GB

So I really don't think it is a problem of my computers performance. Might there somewhere be a buffer collecting some frames before viewing them?

ejlai, 2021-04-21:
> for me, i cant see any latency. try
> 
> ```python
> from djitellopy import tello
> import cv2
> 
> 
> drone = tello.Tello()
> drone.connect()
> 
> while True:
>     img = drone.get_frame_read().frame
>     cv2.imshow('stream', img)
>     cv2.waitKey(1)
> ```
> 
> took it from here, worked fine for me
> https://www.youtube.com/watch?v=LmEcyQnfpDA&t=12027s

My image window won't even show up using the code using Pycharm IDE...not sure what to do next...maybe use IDEL

samlaf, 2021-04-24:
I was also seeing similar latencies with this protocol.
I ended up using the [TelloPy](https://github.com/hanyazou/TelloPy) repo instead, which has reverse engineered the low-level protocol that the official tello application uses. Unfortunately, you can't mix and match both of these protocols, as the tello can only listen on one of the two at any given time. So you will have to choose which repo to use. But TelloPy has nearly 0 video latency.

RalphBariz, 2021-04-25:
Can confirm, with TelloPy there is no latency for me.

closed 2021-06-06 by M4GNV5