damiafuentes/DJITelloPy #89: Did not receive state packet

paulbaumgarten, 2021-05-11:
Hello - Can someone enlighten me of possible causes of the "did not receive a state packet" error? I saw the old/closed issues about this and have turned off Windows Firewall on the laptop in question but still to no avail. Any other suggestions to check? With logger set to DEBUG, the output was...

```
[INFO] tello.py - 106 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[DEBUG] tello.py - 418 - Waiting 0.002994060516357422 seconds to execute command: command...[INFO] tello.py - 421 - Send command: 'command'
[DEBUG] tello.py - 129 - Data received from 192.168.10.1 at client_socket
[INFO] tello.py - 445 - Response command: 'ok'
Traceback (most recent call last):
  File "c:/Users/hoh39/Documents/drones/first.py", line 13, in <module>
    tello.connect()
  File "C:\Users\hoh39\AppData\Roaming\Python\Python38\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "C:\Users\hoh39\AppData\Roaming\Python\Python38\site-packages\djitellopy\tello.py", line 530, in connect
    raise Exception('Did not receive a state packet from the Tello')
Exception: Did not receive a state packet from the Tello
```

Thank you

M4GNV5, 2021-05-13:
Hey,
it looks like you are receiving responses to your command but no state information. Tellos normally send state values every 100ms which include information like battery status, mission pads, flight data etc.
Please make sure you are running the latest Tello firmware (try updating via the official App). Are you sure there is no other antivirus or firewall blocking the communication?
Alternatively you can also try calling [the connect function](https://github.com/damiafuentes/DJITelloPy/blob/master/djitellopy/tello.py#L515) which `False` as its first argument. This simply disables waiting for state packets and you should be able to control your drone fine.

```python
from djitellopy import Tello

tello = Tello()
tello.connect(False) # <- False here

tello.takeoff()
tello.land()
```

paulbaumgarten, 2021-05-21:
I discovered I had missed a setting on the Firewall. Thanks for your help.

closed 2021-05-21 by paulbaumgarten