damiafuentes/DJITelloPy #90: Issues with Tello video stream to Windows Computer

mbrh33, 2021-05-16:
Hi everyone!

I am trying to follow along Murtaza's Workshop YouTube "Drone Programming With Python Course | 3 Hours | Including x4 Projects | Computer Vision" course:

[MEDIA=youtube]LmEcyQnfpDA:2105[/MEDIA]

However I have hit a wall at the  imageCapture section.

The Tello drone connects to the computer and I can use the basic movements commands without any difficulty, having the drone take off / move around with commands issued from the laptop. However when I try to run the code below to get a video stream:

from djitellopy import tello

import cv2

me = tello.Tello()

me.connect()

print(me.get_battery())

me.streamon()

while True:

    img = me.get_frame_read().frame

    img = cv2.resize(img, (360, 240))

    cv2.imshow("Image", img)

    cv2.waitKey(1)
 
the code returns the error:

> [h264 @ 000001d171fc08c0] non-existing PTraceback (most recent call last):
> 
> File "C:/Users/Andrew Rudy/PycharmProjects/TelloDroneYouTubeProject/Video test 2.py", line 11, in <module>
>     frame_read = tello.get_frame_read()
> 
> File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
>     return func(*args, **kwargs)
> 
> File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\tello.py", line 404, in get_frame_read
>     self.background_frame_read = BackgroundFrameRead(self, address)  # also sets self.cap
> 
> File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\tello.py", line 932, in __init__
>     raise Exception('Failed to grab first frame from video stream')

If you debug look at the return value for the Frame = "None" so there is no stream is coming back to the computer. Also seen in the returned "raise Exception('Failed to grab first frame from video stream')".

I have tried to turn the firewall completely off and also to create special inbound rules to allow the UDP port 11111 to come through, however that has not solved the issue.

I am using Windows 10, pycharm, and have tried this with Python 3.7 and 3.9. Additionally I have tried it with djitello and djitello2, am using opencv-python 4.5.2.52.

The issue isn't with the Tello drone - it can send a video to my phone, and my friend ran the same code on his laptop (running on Linux) and it opened up a live stream.

Does anyone have any idea how to troubleshoot this or what may be wrong? Thanks in advance!

M4GNV5, 2021-05-16:
Hey,

this sounds like a problem with your opencv installation. Can you paste the full output of the program? You only included one line of the output ffmpeg produces (see below). But there is probably more output above that line?
> [h264 @ 000001d171fc08c0] non-existing PTraceback (most recent call last):

There seem to be some licensing issues with h264 in opencv, [as described here](https://github.com/opencv/opencv-python/issues/100), but thats just me guessing. Having the full log might help narrowing down your problem.

mbrh33, 2021-05-16:
Thanks for the response - here is the full output:

> [INFO] tello.py - 106 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
> [INFO] tello.py - 421 - Send command: 'command'
> [INFO] tello.py - 445 - Response command: 'ok'
> [INFO] tello.py - 421 - Send command: 'streamon'
> [INFO] tello.py - 445 - Response streamon: 'ok'
> Traceback (most recent call last):
>   File "C:/Users/Andrew Rudy/PycharmProjects/TelloDroneYouTubeProject/Video test 2.py", line 11, in <module>
>     frame_read = tello.get_frame_read()
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
>     return func(*args, **kwargs)
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\tello.py", line 404, in get_frame_read
>     self.background_frame_read = BackgroundFrameRead(self, address)  # also sets self.cap
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\tello.py", line 932, in __init__
>     raise Exception('Failed to grab first frame from video stream')
> Exception: Failed to grab first frame from video stream
> [INFO] tello.py - 421 - Send command: 'streamoff'
> [WARNING] tello.py - 431 - Aborting command 'streamoff'. Did not receive a response after 7 seconds
> [INFO] tello.py - 421 - Send command: 'streamoff'
> [WARNING] tello.py - 431 - Aborting command 'streamoff'. Did not receive a response after 7 seconds
> [INFO] tello.py - 421 - Send command: 'streamoff'
> [WARNING] tello.py - 431 - Aborting command 'streamoff'. Did not receive a response after 7 seconds
> Exception ignored in: <function Tello.__del__ at 0x000001D840DA35E8>
> Traceback (most recent call last):
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\tello.py", line 913, in __del__
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\tello.py", line 902, in end
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\tello.py", line 563, in streamoff
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\tello.py", line 470, in send_control_command
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\tello.py", line 513, in raise_result_error
> Exception: Command 'streamoff' was unsuccessful for 4 tries. Latest response:	'Aborting command 'streamoff'. Did not receive a response after 7 seconds'
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] decode_slice_header error
> [h264 @ 000001d82ffbc100] no frame!
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] decode_slice_header error
> [h264 @ 000001d82ffbc100] no frame!
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] decode_slice_header error
> [h264 @ 000001d82ffbc100] no frame!
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] decode_slice_header error
> [h264 @ 000001d82ffbc100] no frame!
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] decode_slice_header error
> [h264 @ 000001d82ffbc100] no frame!
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] decode_slice_header error
> [h264 @ 000001d82ffbc100] no frame!
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] decode_slice_header error
> [h264 @ 000001d82ffbc100] no frame!
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] non-existing PPS 0 referenced
> [h264 @ 000001d82ffbc100] decode_slice_header error
> [h264 @ 000001d82ffbc100] no frame!
> [h264 @ 000001d840ed6900] non-existing PPS 0 referenced
> [h264 @ 000001d840ed6900] decode_slice_header error
[... 1 lines cut ...]
> [h264 @ 000001d8418038c0] non-existing PPS 0 referenced
> [h264 @ 000001d8418038c0] decode_slice_header error
> [h264 @ 000001d8418038c0] no frame!
> [h264 @ 000001d840f01100] non-existing PPS 0 referenced
> [h264 @ 000001d840f01100] decode_slice_header error
> [h264 @ 000001d840f01100] no frame!
[... cut ...]
> [h264 @ 000001d8417fe740] non-existing PPS 0 referenced
> [h264 @ 000001d8417fe740] decode_slice_header error
> [h264 @ 000001d8417fe740] no frame!
> [h264 @ 000001d840f1bfc0] non-existing PPS 0 referenced
> [h264 @ 000001d840f1bfc0] decode_slice_header error
> [h264 @ 000001d840f1bfc0] no frame!
> [h264 @ 000001d840f1c440] non-existing PPS 0 referenced
> [h264 @ 000001d840f1c440] decode_slice_header error
> [h264 @ 000001d840f1c440] no frame!
> 
> Process finished with exit code 1

Let me know if this helps.

CurlyWurly-1, 2021-05-17:
I had this problem too _(and also another problem where the positive forward values made the drone go backward!)_  

I downgraded opencv-python from **5.5.2.52** to **4.5.1.48** (next version down) and it seems to be OK now

mbrh33, 2021-05-18:
I downgraded to 4.5.1.48, however this didn't work for me. Running the same code posted above produced:

> "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\Scripts\python.exe" "C:/Users/Andrew Rudy/PycharmProjects/TelloDroneYouTubeProject/Video test 2.py"
> [INFO] tello.py - 106 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
> [INFO] tello.py - 421 - Send command: 'command'
> [INFO] tello.py - 445 - Response command: 'ok'
> [INFO] tello.py - 421 - Send command: 'streamon'
> [INFO] tello.py - 445 - Response streamon: 'ok'
> [INFO] tello.py - 421 - Send command: 'streamoff'
> [WARNING] tello.py - 431 - Aborting command 'streamoff'. Did not receive a response after 7 seconds
> [INFO] tello.py - 421 - Send command: 'streamoff'
> [WARNING] tello.py - 431 - Aborting command 'streamoff'. Did not receive a response after 7 seconds
> [INFO] tello.py - 421 - Send command: 'streamoff'
> [WARNING] tello.py - 431 - Aborting command 'streamoff'. Did not receive a response after 7 seconds
> Exception ignored in: <function Tello.__del__ at 0x000002526E1BD5E8>
> Traceback (most recent call last):
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\tello.py", line 913, in __del__
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\tello.py", line 902, in end
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\tello.py", line 563, in streamoff
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\tello.py", line 470, in send_control_command
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\enforce_types.py", line 54, in wrapper
>   File "C:\Users\Andrew Rudy\PycharmProjects\TelloDroneYouTubeProject\venv\lib\site-packages\djitellopy\tello.py", line 513, in raise_result_error
> Exception: Command 'streamoff' was unsuccessful for 4 tries. Latest response:	'Aborting command 'streamoff'. Did not receive a response after 7 seconds'
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] decode_slice_header error
> [h264 @ 000002526e2ee380] no frame!
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] decode_slice_header error
> [h264 @ 000002526e2ee380] no frame!
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] decode_slice_header error
> [h264 @ 000002526e2ee380] no frame!
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] decode_slice_header error
> [h264 @ 000002526e2ee380] no frame!
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] decode_slice_header error
> [h264 @ 000002526e2ee380] no frame!
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] decode_slice_header error
> [h264 @ 000002526e2ee380] no frame!
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] decode_slice_header error
> [h264 @ 000002526e2ee380] no frame!
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] decode_slice_header error
> [h264 @ 000002526e2ee380] no frame!
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] decode_slice_header error
> [h264 @ 000002526e2ee380] no frame!
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] non-existing PPS 0 referenced
> [h264 @ 000002526e2ee380] decode_slice_header error
> [h264 @ 000002526e2ee380] no frame!
> [h264 @ 000002526e2ace00] non-existing PPS 0 referenced
> [h264 @ 000002526e2ace00] decode_slice_header error
> [h264 @ 000002526e2ace00] no frame!
[... 6 lines cut ...]
> [h264 @ 000002526edcf980] non-existing PPS 0 referenced
> [h264 @ 000002526edcf980] decode_slice_header error
> [h264 @ 000002526edcf980] no frame!
> [h264 @ 000002526e2efd80] non-existing PPS 0 referenced
> [h264 @ 000002526e2efd80] decode_slice_header error
> [h264 @ 000002526e2efd80] no frame!
> [h264 @ 000002526e89dc00] non-existing PPS 0 referenced
> [h264 @ 000002526e89dc00] decode_slice_header error
> [h264 @ 000002526e89dc00] no frame!
> [h264 @ 000002526ea700c0] non-existing PPS 0 referenced
[... cut ...]
> [h264 @ 000002526ea70540] non-existing PPS 0 referenced
> [h264 @ 000002526ea70540] decode_slice_header error
> [h264 @ 000002526ea70540] no frame!
> [h264 @ 000002526e2ace00] non-existing PPS 0 referenced
> [h264 @ 000002526e2ace00] decode_slice_header error
> [h264 @ 000002526e2ace00] no frame!
> [h264 @ 000002526e911780] non-existing PPS 0 referenced
> [h264 @ 000002526e911780] decode_slice_header error
> [h264 @ 000002526e911780] no frame!
> 
> Process finished with exit code 0

Any other ideas on what might potentially work?

CurlyWurly-1, 2021-05-18:
In Pycharm->Settings, These are the combinations of settings which seem to work for Murtazza and myself
				Murtazza 		My setupy
Python Interpretor	3.7			3.7

djitellopy			1.5			1.5
numpy			1.19.0		1.20.3
opencv-python		4.3.0.36		4.5.1.48
pip				19.0.3		21.1.1
pygame			1.9.6			2.0.1
setuptools		40.8.0		56.2.0`

CurlyWurly-1, 2021-05-18:
What code are you using? - All the code for Murtazza seemsto work for me (after a bit of tweaking)

mbrh33, 2021-05-19:
I have the settings:
Pillow | 8.2.0 | 8.2.0
cycler | 0.10.0 | 0.10.0
djitellopy2 | 2.3 | 2.3
kiwisolver | 1.3.1 | 1.3.1
matplotlib | 3.4.2 | 3.4.2
numpy | 1.20.3 | 1.20.3
opencv-python | 4.5.1.48 | 4.5.2.52
pip | 21.1.1 | 21.1.1
pygame | 2.0.1 | 2.0.1
pyparsing | 2.4.7 | 2.4.7
python-dateutil | 2.8.1 | 2.8.1
setuptools | 56.1.0 | 56.2.0
six | 1.16.0 | 1.16.0

I have been trying to trouble shoot with other posted code to get video streaming on the tello to work, which is why there are extra packages installed. I've tried djitellopy 1.5 as well and that hasn't worked. 

The code I'm using is outlined is the same as Murtazza's - it is in my first post. 

Any other ideas on a fix? How did you tweak his code?

CurlyWurly-1, 2021-05-19:
One difference you have - try uninstalling Djitellopy2 and then installing Djitellopy?
The code you posted is the same really - I had to delete and retype the quote marks - all done in pycharm.
It does sound like an environment issue - you could try making a project with its own environment, and only install the ones shown in my list

uqyge, 2021-06-03:
had the same issue and **opencv-python 4.5.1.48** saved it.

> In Pycharm->Settings, These are the combinations of settings for Murtazza and myself
> 
> **Murtazza Settings**
> Python Interpretor 3.7
> djitellopy 1.5
> numpy 1.19.0
> opencv-python 4.3.0.36
> pip 19.0.3
> pygame 1.9.6
> setuptools 40.8.0 `
> 
> **My settings**
> Python Interpretor 3.7
> djitellopy 1.5
> numpy 1.20.3
> opencv-python 4.5.1.48
> pip 21.1.1
> pygame 2.0.1
> setuptools 56.2.0

mbrh33, 2021-06-06:
So I got it to work by changing the tello.py file 

https://github.com/damiafuentes/DJITelloPy/blob/aad98c1e8d8fd916112e7e52eba1318398adb6ad/djitellopy/tello.py#L930

in line 930 I changed the if statement to a while statement for grabbing the first frame. It appears that my computer took longer to initialize that first frame and the exception would cancel out the code.   

```
        self.grabbed, self.frame = self.cap.read()
        while not self.grabbed or self.frame is None:
            print('still trying to grab first frame')
            self.grabbed, self.frame = self.cap.read()
```

This gave it time to be able to grab the first frame and initialize. Going to see if I can find a workaround that doesn't involve editing the tello.py file.  

If you look around online it seems that a lot of people are having this issue so it might be nice if the base file could be updated.

M4GNV5, 2021-06-06:
Hmm interesting. I guess its a good idea to merge this adjustment into upstream then.
However some sort of timeout is required here. With your code it just loops infinitely if something is actually broken. A better approach would be to try `self.cap.read()` 5-10 times or until `X` seconds passed.

M4GNV5, 2021-06-06:
I've implemented this with trying 20 times to grab a frame with a 0.05s delay. Can you confirm wether this fixes your issue @mbrh33

closed 2021-06-06 by M4GNV5

zelvet-velvet, 2022-05-12:
> I've implemented this with trying 20 times to grab a frame with a 0.05s delay. Can you confirm wether this fixes your issue @mbrh33?

I have tried the the way you suggested but it still throw the exception "Failed to grab first frame from video stream".
Below is the output of execute:
↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓
[INFO] tello.py - 107 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 422 - Send command: 'command'
[INFO] tello.py - 446 - Response command: 'ok'
[INFO] tello.py - 422 - Send command: 'port 8890 11111'
[INFO] tello.py - 446 - Response port 8890 11111: 'ok'
[INFO] tello.py - 422 - Send command: 'streamon'
[INFO] tello.py - 446 - Response streamon: 'ok'
Traceback (most recent call last):
  File "stream_test.py", line 13, in <module>
    img=tello.get_frame_read().frame
  File "/home/zel/anaconda3/envs/tello/lib/python3.6/site-packages/djitellopy/enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "/home/zel/anaconda3/envs/tello/lib/python3.6/site-packages/djitellopy/tello.py", line 405, in get_frame_read
    self.background_frame_read = BackgroundFrameRead(self, address)  # also sets self.cap
  File "/home/zel/anaconda3/envs/tello/lib/python3.6/site-packages/djitellopy/tello.py", line 943, in __init__
    raise Exception('Failed to grab first frame from video stream')
Exception: Failed to grab first frame from video stream 
↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑

My code is here:
↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓↓
import cv2
import time
from djitellopy import Tello

tello = Tello()
tello.connect(False)

tello.send_command_with_return("port 8890 11111")

tello.streamon()
while True:
	img=tello.get_frame_read()
	cv2.imshow("Image",img)
	cv2.waitKey(1)

tello.land()
↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑↑
Does any one have any idea how to solve this.
Thanks.

M4GNV5, 2022-05-12:
Hey @zelvet-velvet,

your problem looks like a different issue. Maybe you want to open a separate issue for it?
Also why are you using the `port` command instead of using the [corresponding function](https://github.com/damiafuentes/DJITelloPy/blob/master/djitellopy/tello.py#L836). Also you probably do not need to use that function at all.

Furthermore based on your exception the code you sent is not the code you are actually using.

zelvet-velvet, 2022-05-19:
> Hey @zelvet-velvet,
>
> your problem looks like a different issue. Maybe you want to open a separate issue for it? Also why are you using the `port` command instead of using the [corresponding [...]

Hi @M4GNV5 ,
I have updated my issue in #153 . Since tello.py have been update, there occured an another error about av. Hope to see you there. Thanks.

commit 0b478a04 (2021-06-06): try grabbing the first frame multiple times giving the decoder some time to become ready (fix #90)