damiafuentes/DJITelloPy #91: Connection Issue - Did not receive a state packet from the Tello

RushilJoshi, 2021-06-04:
I'm trying to connect to the tello using:

```
from djitellopy import tello

me = tello.Tello()
me.connect()
print(me.battery())
```
I am getting the following error readout:

```
[INFO] tello.py - 106 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 421 - Send command: 'command'
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.6/3.6.12_1/Frameworks/Python.framework/Versions/3.6/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python@3.6/3.6.12_1/Frameworks/Python.framework/Versions/3.6/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.6/site-packages/djitellopy/enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/djitellopy/tello.py", line 148, in udp_state_receiver
    state_socket.bind(("", Tello.STATE_UDP_PORT))
OSError: [Errno 48] Address already in use

[INFO] tello.py - 445 - Response command: 'ok'
Traceback (most recent call last):
  File "tello_camera.py", line 9, in <module>
    me.connect()
  File "/usr/local/lib/python3.6/site-packages/djitellopy/enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/djitellopy/tello.py", line 530, in connect
    raise Exception('Did not receive a state packet from the Tello')
Exception: Did not receive a state packet from the Tello
```

I am on a macbook pro with Big Sur, using Python 3.6.12.  What could be causing this?

I have attempted pasting the following into /etc/pf.conf

```
pass in proto udp from any to any port 11111
pass in proto udp from any to any port 8889
pass in proto udp from any to any port 8890
```

and then running
```
sudo pfctl -d
sudo pfctl -E
```

with no luck.

M4GNV5, 2021-06-06:
Hey,
your error message `OSError: [Errno 48] Address already in use` means something is already listening on port `8890`, i.e. another program is already running which uses the port Tellos use for state packets. You can list programs using `netstat -ulpen` to check which program uses port `8890`.
Are you trying to run your script twice? Or is it some other program?

marcelovidigal, 2021-06-19:
I have the same problem, I tested the code on windows and it worked perfectly, but on macosx this problem appears (Did not receive a state packet from the Tello), in my case I totally disabled the firewall (solution presented here) and still the error persists. What can this be? Any advances specifically for macosx?

M4GNV5, 2021-06-20:
Can you post your error message? Do you also get the `Address already in use` error? I described above the problem which causes this.

marcelovidigal, 2021-06-20:
The error is the title message (Exception: Did not receive a state packet from the Tello), the same as other posts here, I believe it only happens on mac, and there was no message that the 'Address already in use'.

marcelovidigal, 2021-06-24:
Testing with Python 3.8 worked perfectly, either using tello.connect(False) or tello.connect(), as I was using Python 3.9 I'm referring to some problem with Python versioning.

```
from djitellopy import Tello

tello = Tello()

tello.connect(False)
or
tello.connect()

print(tello.get_battery())

tello.takeoff()
tello.land()
```

The downgrade to version 3.8 solved the problem for me, although not ideal, it is now possible to control the drone of a mac osx big sur computer.

Thanks for the help.

closed 2021-06-24 by M4GNV5

mgrallos, 2022-10-21:
Hi i am using an Ubuntu 20.04 VM 3.8 python and firewall disabled but the same error persists. Send help

lafkpages, 2024-05-20:
I can reproduce this issue using Python 3.12.3 on macOS Sonoma 14.3.1. I do get an "address already in use" error, but I checked using `lsof` and before running the below code, nothing is listening on neither `8889` or `8890`. While I run the code, it listens on `8889`, and once it crashes there is again nothing listening on either of the ports.

Code:
```py
from djitellopy import Tello
from time import sleep

tello = Tello()

tello.connect()
tello.takeoff()

sleep(5)

tello.land()
```

Output:
```
[INFO] tello.py - 129 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
Exception in thread Thread-2 (udp_state_receiver):
Traceback (most recent call last):
  File "/opt/homebrew/Cellar/python@3.12/3.12.3/Frameworks/Python.framework/Versions/3.12/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/opt/homebrew/Cellar/python@3.12/3.12.3/Frameworks/Python.framework/Versions/3.12/lib/python3.12/threading.py", line 1010, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/luis/Documents/coding/dji-tello-test/lib/python3.12/site-packages/djitellopy/enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/luis/Documents/coding/dji-tello-test/lib/python3.12/site-packages/djitellopy/tello.py", line 180, in udp_state_receiver
    state_socket.bind(("", Tello.STATE_UDP_PORT))
OSError: [Errno 48] Address already in use
[INFO] tello.py - 438 - Send command: 'command'
[INFO] tello.py - 462 - Response command: 'ok'
Traceback (most recent call last):
  File "/Users/luis/Documents/coding/dji-tello-test/main.py", line 6, in <module>
    tello.connect()
  File "/Users/luis/Documents/coding/dji-tello-test/lib/python3.12/site-packages/djitellopy/enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/luis/Documents/coding/dji-tello-test/lib/python3.12/site-packages/djitellopy/tello.py", line 547, in connect
    raise TelloException('Did not receive a state packet from the Tello')
djitellopy.tello.TelloException: Did not receive a state packet from the Tello
```