damiafuentes/DJITelloPy #92: Add support for new commands introduced in SDK 3.0

M4GNV5, 2021-06-12:
A new Tello firmware updates brings new commands which are currently missing in DJITelloPy. See [SDK 3.0 Documentation](https://dl.djicdn.com/downloads/RoboMaster+TT/Tello_SDK_3.0_User_Guide_en.pdf) and [this TelloPilots forum thread](https://tellopilots.com/threads/sdk-3-0-for-tello-edu.6013/#post-38489)

Commands to implement:
- [ ] `motoroff`
- [ ] `motoron`
- [ ] `port`
- [ ] `reboot`
- [ ] `setbitrate`
- [ ] `setfps`
- [ ] `setresolution`
- [ ] `throwfly`
- [ ] `ssid?`
- [ ] `hardware?`

2black0, 2021-06-27:
wow its great news, tello can set be station mode and connect to AP, so this will unlock for tello (not edu) do swarm flying

M4GNV5, 2021-09-01:
The following additional commands were discovered by users in [this thread](https://tellopilots.com/threads/sdk-3-0-for-tello-edu.6013/page-2):

- [ ] `active?`
- [ ] `keepalive`
- [ ] `downvision`

M4GNV5, 2021-09-05:
All commands which do not require a Tello TT are implemented. I will close this issue. Progress of TTs "open source controller" can/should be tracked in a sperate issue.

closed 2021-09-05 by M4GNV5