damiafuentes/DJITelloPy #93: connect_to_wifi not rebooting

h3retek, 2021-06-17:
So, following this:

```
from djitellopy import *
import time

tello = Tello()

if __name__ == '__main__':
    tello.connect()
    time.sleep(2)
    tello.connect_to_wifi('ssid', 'passwd')
```

the Tello does not reboot... it just sits there.

I can send it commands, and it works... but I cannot get it to connect to my home network

Firmware: 1.04.91.01
NOTE: If I try to update firmware to 2.05.01.17 the app says that the firmware is already on the Latest version

M4GNV5, 2021-06-17:
Are you using a regular Tello? Or an Tello EDU?
Only EDU Tellos are capable of connecting to an wifi access point.

h3retek, 2021-06-18:
Thank you sir... That would be the reason. T~T

closed 2021-06-18 by h3retek