damiafuentes/DJITelloPy #95: Exception: Did not receive a state packet from the Tello on MacOSX

marcelovidigal, 2021-06-19:
In macosx big sur this problem prevents the execution of any code, always with this message. What can this be? Has anyone experiencing this problem been able to work around this? The machine's firewall is totally disabled. It appears to be a mac UDP issue, according to other reports. Thanks for the help, I tested a simple example on windows and it worked, but on macosx, the machine I use, I can't go forward because of this problem.

M4GNV5, 2021-06-20:
Please post the full error message you get when running your python program as well a python program which can be used to reconstruct your error.

Maybe a duplicate of #91 and/or #71 / #79?

marcelovidigal, 2021-06-20:
> Please post the full error message you get when running your python program as well a minimal python program which can be used to reproduce your error.
> 
> Maybe a duplicate of #91 and/or #71 / #79?

The error is the title message (Exception: Did not receive a state packet from the Tello), the same as other posts here, I believe it only happens on mac, and there was no message that the connection is already being used.

M4GNV5, 2021-06-20:
Please post the full output of your program. Since you already posted at #91 did you also receive a `Address already in use` error? Or only the `Did not receive a state packet` error?
Does your Tello work when using the code snippet here? https://github.com/damiafuentes/DJITelloPy/issues/89#issuecomment-840514577

marcelovidigal, 2021-06-20:
No 'Address already in use error' message. This is a simple example of battery check, the only output is this message error 'Exception: Did not receive a state packet from the Tello'. Firewall is completely disabled. OS is OSX Big Sur.

`
[INFO] tello.py - 107 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 422 - Send command: 'command'
[INFO] tello.py - 446 - Response command: 'ok'
Traceback (most recent call last):
File "/dji/tello/programa_tello.py", line 12, in <module>
    tello.connect()
File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/djitellopy/enforce_types.py", line 54, in wrapper
    return func(*args, **kwargs)
File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/djitellopy/tello.py", line 531, in connect
    raise Exception('Did not receive a state packet from the Tello')
Exception: Did not receive a state packet from the Tello`

marcelovidigal, 2021-06-24:
Testing with Python 3.8 worked perfectly, either using tello.connect(False) or tello.connect(), as I was using Python 3.9 I'm referring to some problem with Python versioning.

```
from djitellopy import Tello

tello = Tello()

tello.connect(False)
or
tello.connect()

print(tello.get_battery())

tello.takeoff()
tello.land()
```

The downgrade to version 3.8 solved the problem for me, although not ideal, it is now possible to control the drone of a mac osx big sur computer.

Thanks for the help.