damiafuentes/DJITelloPy #96: Tello not reporting correct X-Velocity

VaeZarek, 2021-06-22:
Hello, my Tello drone does not seem to be reporting a correct velocity in the x direction. My code looks like:

from djitellopy import Tello
from threading import Thread
import time

droneVelocity = 0

def speedometer():
    seconds = 0
    while tello.is_flying == True:
        droneVelocity = tello.get_speed_x()
        print("Velocity of Drone: " + str(droneVelocity))
        print("At " + str(seconds) + " seconds")
        seconds += 1/10
        time.sleep(1/10)

speedCheck = Thread(target=speedometer)
tello = Tello()

tello.connect()
print(tello.get_battery())
tello.takeoff()
speedCheck.start()
tello.move_forward(100)
tello.move_back(100)
tello.land()

From this, the drone moves forward 100 centimeters and back 100 centimeters. These both take 2 seconds, so the drone x velocity should be 50 cm/s, but the max it is reporting is around 7 cm/s. This makes sense in decimeters but that is a weird unit of measurement. Is there something else wrong with my code or that I should try? Thanks.

M4GNV5, 2021-06-22:
Hey,
can you paste the full output of the program?
My first guess is the drone tilts itself in the direction it wants to fly and thus the actual speed is divided in a X and Z component.
Did you also try printing the other speeds (Y and Z)?

VaeZarek, 2021-06-22:
I have not tried the other speeds yet, but I will implement them and generate an output. For the code I posted before the output is:

[INFO] tello.py - 107 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 422 - Send command: 'command'
[INFO] tello.py - 446 - Response command: 'ok'
50
[INFO] tello.py - 422 - Send command: 'takeoff'
[INFO] tello.py - 446 - Response takeoff: 'ok'
Velocity of Drone: 0
At 0 seconds
[INFO] tello.py - 422 - Send command: 'forward 100'
Velocity of Drone: 0
At 0.1 seconds
Velocity of Drone: 0
At 0.2 seconds
Velocity of Drone: 0
At 0.30000000000000004 seconds
Velocity of Drone: 0
At 0.4 seconds
Velocity of Drone: -3
At 0.5 seconds
Velocity of Drone: -4
At 0.6 seconds
Velocity of Drone: -5
At 0.7 seconds
Velocity of Drone: -6
At 0.7999999999999999 seconds
Velocity of Drone: -7
At 0.8999999999999999 seconds
Velocity of Drone: -7
At 0.9999999999999999 seconds
Velocity of Drone: -7
At 1.0999999999999999 seconds
Velocity of Drone: -6
At 1.2 seconds
Velocity of Drone: -5
At 1.3 seconds
Velocity of Drone: -4
At 1.4000000000000001 seconds
Velocity of Drone: -3
At 1.5000000000000002 seconds
Velocity of Drone: -3
At 1.6000000000000003 seconds
Velocity of Drone: -3
At 1.7000000000000004 seconds
Velocity of Drone: -3
At 1.8000000000000005 seconds
Velocity of Drone: -2
At 1.9000000000000006 seconds
Velocity of Drone: -2
At 2.0000000000000004 seconds
Velocity of Drone: -2
At 2.1000000000000005 seconds
Velocity of Drone: -2
At 2.2000000000000006 seconds
Velocity of Drone: -2
At 2.3000000000000007 seconds
Velocity of Drone: -1
At 2.400000000000001 seconds
Velocity of Drone: -1
At 2.500000000000001 seconds
Velocity of Drone: 0
At 2.600000000000001 seconds
Velocity of Drone: 0
At 2.700000000000001 seconds
Velocity of Drone: 0
At 2.800000000000001 seconds
[INFO] tello.py - 446 - Response forward 100: 'ok'
[INFO] tello.py - 422 - Send command: 'back 100'
Velocity of Drone: 0
[... 420 lines cut ...]
X Velocity of Drone: 0
Y Velocity of Drone: 0
Z Velocity of Drone: -4
At 6.799999999999992 seconds
X Velocity of Drone: 0
Y Velocity of Drone: 0
Z Velocity of Drone: -4
At 6.8999999999999915 seconds
X Velocity of Drone: 0
Y Velocity of Drone: 0
Z Velocity of Drone: -4
At 6.999999999999991 seconds
X Velocity of Drone: 0
Y Velocity of Drone: 0
Z Velocity of Drone: -3
At 7.099999999999991 seconds
X Velocity of Drone: 0
Y Velocity of Drone: 0
Z Velocity of Drone: -3
At 7.19999999999999 seconds
X Velocity of Drone: 0
Y Velocity of Drone: 0
Z Velocity of Drone: -2
At 7.29999999999999 seconds
[INFO] tello.py - 446 - Response land: 'ok'

AbdullahAmer86, 2021-07-10:
Hello I am getting a similar issue, I reduced the code to a very simple one and the issue is the same, 

[INFO] tello.py - 446 - Response takeoff: 'ok'
[INFO] tello.py - 455 - Send command (no response expected): 'rc 0 20 0 0'
[INFO] tello.py - 422 - Send command: 'land'
[WARNING] tello.py - 432 - Aborting command 'land'. Did not receive a response after 7 seconds
[INFO] tello.py - 422 - Send command: 'land'
[INFO] tello.py - 446 - Response land: 'ok'

There no response to the rc control command. any idea what is the issue?

M4GNV5, 2021-07-11:
@AbdullahAmer86 I do not think the output you posted is related to this issue. This issue is about the speed reported by the Tello in state packets. Your output looks fine, the `rc` command does not send a response and is not expected to do though. Even though your `land` fails it succeeds on the second try, so that looks fine too.

@VaeZarek the incorrect speed is either a problem with your specific Tello or a bug in the Tello firmware. I will do some testing with one of my Tellos. If it shows the same incorrect behaviour there we should contact Ryze support, otherwise your Tello might be broken or something.

AbdullahAmer86, 2021-07-11:
@M4GNV5 thank you for your comments, but I don't understand the sentence saying (no response expected)?

M4GNV5, 2021-07-11:
The [send_rc_control](https://github.com/damiafuentes/DJITelloPy/blob/master/djitellopy/tello.py#L773) function uses [send_command_without_return](https://github.com/damiafuentes/DJITelloPy/blob/master/djitellopy/tello.py#L449). This means the command is simply sent there but the Tello does not send a response like `ok` back.
The `rc` command is often used e.g. when tracking an object in the received video stream. When you for example want to follow a moving object you continuously calculate the required drone speed and send it e.g. 10 times a second using `send_rc_control`. It does not matter if one of 10 packets is lost as there are so many which constantly set the desired drone speed, thus there is no response (i.e. no `ok` or `error`).

AbdullahAmer86, 2021-07-13:
@M4GNV5 Thanks a lot for you clear explanation. I am learning using your library and I am quite thankful for all the efforts.

M4GNV5, 2021-11-13:
@VaeZarek i finally found some time and a Tello to try your code myself. In fact the velocity seems to be between 0 and 10. As i guessed when moving forward the drone tilts and the velocity is split into a larger X and smaller Y direction. For me the maximum speed is

```
At 1.0999999999999999 seconds
X Velocity of Drone: 7
Y Velocity of Drone: 3
Z Velocity of Drone: 0
```

Which would add up to a little bit less than 10 in the "forward" direction. With the speed set to 100cm/s i assume the speed is reported in dm/s just as you assumed.

So i assume everything works as intended? At least on my end, your drone does not seem to provide a Y velocity. What firmware are you running?

VaeZarek, 2021-11-13:
Hello, I don't currently have access to my drones for the time being, so I can't look up what firmware I updated them to. I did the latest update on the Tello Android app as of July 31st, 2021, and it looks like that would be 02.05.01.17

I hope this helps, I will check when I get access to the drones again. Thank You.