damiafuentes/DJITelloPy #98: Robot master TT does not provide

FachengG, 2021-07-05:
I just get an Robot master TT for its off-line program feature. I am very new to drone. 

I encounter a problem with the camera (take picture / video stream), here is a bug example:

When execute following code:
```
import cv2
import time
from djitellopy import Tello

tello = Tello()
tello.connect()

time.sleep(1)
tello.takeoff()
tello.move_forward(100)
tello.streamon()
frame_read = tello.get_frame_read()
cv2.imwrite("picture.png", frame_read.frame)

tello.land()
```

Here is my log:
```
[INFO] tello.py - 107 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 422 - Send command: 'command'
[INFO] tello.py - 446 - Response command: 'ok'
[INFO] tello.py - 422 - Send command: 'takeoff'
[INFO] tello.py - 446 - Response takeoff: 'ok'
[INFO] tello.py - 422 - Send command: 'forward 100'
[INFO] tello.py - 446 - Response forward 100: 'ok'
[INFO] tello.py - 422 - Send command: 'streamon'
[INFO] tello.py - 446 - Response streamon: 'ok'
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] non-existing PPS 0 referenced
[h264 @ 000001f5f4cec340] decode_slice_header error
[h264 @ 000001f5f4cec340] no frame!
[h264 @ 000001f5ffad1180] non-existing PPS 0 referenced
[h264 @ 000001f5ffad1180] decode_slice_header error
[h264 @ 000001f5ffad1180] no frame!
[h264 @ 000001f586709980] non-existing PPS 0 referenced
[h264 @ 000001f586709980] decode_slice_header error
[h264 @ 000001f586709980] no frame!
[h264 @ 000001f58673f700] non-existing PPS 0 referenced
[h264 @ 000001f58673f700] decode_slice_header error
[h264 @ 000001f58673f700] no frame!
[h264 @ 000001f5ffd7c700] non-existing PPS 0 referenced
[h264 @ 000001f5ffd7c700] decode_slice_header error
[h264 @ 000001f5ffd7c700] no frame!
[h264 @ 000001f5f4ced5c0] non-existing PPS 0 referenced
[h264 @ 000001f5f4ced5c0] decode_slice_header error
[h264 @ 000001f5f4ced5c0] no frame!
[h264 @ 000001f5f4ceda40] non-existing PPS 0 referenced
[h264 @ 000001f5f4ceda40] decode_slice_header error
[h264 @ 000001f5f4ceda40] no frame!
[h264 @ 000001f5ffaea780] non-existing PPS 0 referenced
[h264 @ 000001f5ffaea780] decode_slice_header error
[h264 @ 000001f5ffaea780] no frame!
[h264 @ 000001f5ffaeff00] non-existing PPS 0 referenced
[h264 @ 000001f5ffaeff00] decode_slice_header error
[h264 @ 000001f5ffaeff00] no frame!
[h264 @ 000001f5ffad1180] non-existing PPS 0 referenced
[h264 @ 000001f5ffad1180] decode_slice_header error
[h264 @ 000001f5ffad1180] no frame!
[h264 @ 000001f586709980] non-existing PPS 0 referenced
[h264 @ 000001f586709980] decode_slice_header error
[h264 @ 000001f586709980] no frame!
[h264 @ 000001f58673f700] non-existing PPS 0 referenced
[h264 @ 000001f58673f700] decode_slice_header error
[h264 @ 000001f58673f700] no frame!
[h264 @ 000001f5ffd7c700] non-existing PPS 0 referenced
[h264 @ 000001f5ffd7c700] decode_slice_header error
[h264 @ 000001f5ffd7c700] no frame!
[h264 @ 000001f5f4ced5c0] non-existing PPS 0 referenced
[h264 @ 000001f5f4ced5c0] decode_slice_header error
[h264 @ 000001f5f4ced5c0] no frame!
[h264 @ 000001f5f4ceda40] non-existing PPS 0 referenced
[h264 @ 000001f5f4ceda40] decode_slice_header error
[h264 @ 000001f5f4ceda40] no frame!
[h264 @ 000001f5ffaea780] non-existing PPS 0 referenced
[h264 @ 000001f5ffaea780] decode_slice_header error
[h264 @ 000001f5ffaea780] no frame!
[h264 @ 000001f5ffaeff00] non-existing PPS 0 referenced
[h264 @ 000001f5ffaeff00] decode_slice_header error
[h264 @ 000001f5ffaeff00] no frame!
[h264 @ 000001f5ffad1180] non-existing PPS 0 referenced
[h264 @ 000001f5ffad1180] decode_slice_header error
[h264 @ 000001f5ffad1180] no frame!
[h264 @ 000001f586709980] non-existing PPS 0 referenced
[h264 @ 000001f586709980] decode_slice_header error
[h264 @ 000001f586709980] no frame!
[h264 @ 000001f58673f700] non-existing PPS 0 referenced
[h264 @ 000001f58673f700] decode_slice_header error
[h264 @ 000001f58673f700] no frame!
[h264 @ 000001f5ffd7c700] non-existing PPS 0 referenced
[h264 @ 000001f5ffd7c700] decode_slice_header error
[h264 @ 000001f5ffd7c700] no frame!
[h264 @ 000001f5f4ced5c0] non-existing PPS 0 referenced
[h264 @ 000001f5f4ced5c0] decode_slice_header error
[h264 @ 000001f5f4ced5c0] no frame!
[h264 @ 000001f5f4ceda40] non-existing PPS 0 referenced
[h264 @ 000001f5f4ceda40] decode_slice_header error
[h264 @ 000001f5f4ceda40] no frame!
[INFO] tello.py - 422 - Send command: 'land'
[INFO] tello.py - 446 - Response land: 'ok'
[INFO] tello.py - 422 - Send command: 'streamoff'
[WARNING] tello.py - 432 - Aborting command 'streamoff'. Did not receive a response after 7 seconds
[INFO] tello.py - 422 - Send command: 'streamoff'
```

Could anyone help me with this problem?

M4GNV5, 2021-07-05:
Hey,

your problem sounds similar to #90. Which opencv version are you using?

Also you can try changing your code to the following, basically setting the ports and turning on streaming before takeoff:
```python3
import cv2
import time
from djitellopy import Tello

tello = Tello()
tello.connect()

tello.send_command_with_return("port 8890 11111")

tello.streamon()
frame_read = tello.get_frame_read()

time.sleep(1)

tello.takeoff()
tello.move_forward(100)
cv2.imwrite("picture.png", frame_read.frame)

tello.land()
```

FachengG, 2021-07-06:
Hey M4GNV5, thank you very much for your help!

Unfortunately, I cannot try your solution anymore. I just returned my Robot master TT, because the off-line program system is too weak. It does not worth the money. 

If anyone encounter the same problem with Robot master TT, please try M4GNV5's solution! Hope it will work.

closed 2021-07-06 by FachengG