damiafuentes/DJITelloPy #99: Long OpenCV Buildwheel When Downloading djitellopy package Pycharm

RiqCodedIt, 2021-07-13:
Hi! I am using pycharm to code tello edu's for a summer camp I work at. I was trying to install djitellopy2 on one of the laptops that the kids use and I just get a description in the background task that says Building wheel for opencv-python. This building wheel takes extremely long and the djitellopy2 package never finishes. I tried downloading the djitellopy2 by using pip, but that brings me to the same building wheel for djitellopy2 problem. I am using the latest download of pycharm and Python 3.7. Can someone please tell me what the issue is? Thanks
[image]

M4GNV5, 2021-08-01:
The `djitellopy` package is now the same as the `djitellopy2` so you can pick whichever you like.

Can you post the output of pip when running it? Make sure to add `-v` and `--log` to pip, [as described here](https://stackoverflow.com/questions/20771148/how-can-i-get-pip-installs-download-progress#42234663) which might reveal some additional information.

M4GNV5, 2021-09-05:
I'll close this as i didnt hear back from you. Feel free to repone or open a new issue if the issue still exists.

closed 2021-09-05 by M4GNV5