damiafuentes/DJITelloPy #101: Problems with some examples

Helge543, 2021-07-20:
Hey,

I'm totally new to DJITelloPy or drones in general. Right now I'm trying to do some of the examples which are provided to start. More specifically, I tried `manual-control-pygame.py` and `simple.py.`

When I used to try `simple.py` I wrote a main function like this:

```
from djitellopy import Tello

def main():
    tello = Tello()

    tello.connect()
    tello.takeoff()

    tello.move_left(5)
    tello.rotate_clockwise(90)
    tello.move_forward(5)

    tello.land()

if __name__ == '__main__':
    main()
```

but when I tried to run the program I got the following error message:

```
[INFO] tello.py - 107 - Tello instance was initialized. Host: '192.168.10.1'. Port: '8889'.
[INFO] tello.py - 422 - Send command: 'command'
[INFO] tello.py - 446 - Response command: 'ok'
[INFO] tello.py - 422 - Send command: 'takeoff'
[INFO] tello.py - 446 - Response takeoff: 'error'
[INFO] tello.py - 422 - Send command: 'takeoff'
[INFO] tello.py - 446 - Response takeoff: 'error'
[INFO] tello.py - 422 - Send command: 'takeoff'
[INFO] tello.py - 446 - Response takeoff: 'error'
Traceback (most recent call last):
...
```

It seems that there is a general problem with the `takeoff() `command because when I tried the `manual-control-pygame.py `example some things seem to work fine (connection worked well and I got images from the camera on a separate window) but also here, something seems to be wrong with `takeoff()` command.

Help would be very much appreciated!

Helge543, 2021-07-20:
EDIT: I tried a small delay with `time.sleep(8)` after the `tello.takeoff()` command

That gave me:

```
[INFO] tello.py - 422 - Send command: 'takeoff'
[INFO] tello.py - 446 - Response takeoff: 'ok'
```

But, however the next command after the takeoff throws the next error.

M4GNV5, 2021-07-20:
`tello.move_left(5)` is an invalid command.
The minimum amount a tello can travel is `50 cm` i.e. `tello.move_left(50)` is the minimum possible.

Apart from that try to turn on the light, the tello takeoff command takes longer in bad lighting or wind conditions.

closed 2021-08-01 by M4GNV5